Ultimaker/Cura #1671: Cura doesn't support single nozzle printing on Ultimaker 3

seantapscott, 2017-04-11:
Refuses to let me print without two print cores inserted, Cura does not recognize when a material has been unloaded and left out.

nallath, 2017-04-11:
This is something that the machine wants, so it has little to do with Cura.

You are right that an empty material is not correctly identified by Cura.

seantapscott, 2017-04-11:
It has a lot to do with Cura.  I cannot tell Cura to turn off or not slice with one of the extruders.  The Ultimaker is just reading GCode most of the time.

i.e. The Ultimaker 3 reads out "No Material" when looking at printcore 1, but I cannot select "No material" in Cura.

nallath, 2017-04-11:
The printer as far as I know won't print with one print core. So that part has to do with the printer, whether you agree or not.

seantapscott, 2017-04-11:
Chicken before the egg.

seantapscott, 2017-04-11:
Nallath,

Thank you for the reply.  Just wanted to report it as something that could possibly be improved.  I thought this could be a place for that.  If you feel this should be closed, I'd be fine with it. :)

Thank you for your continued work on Cura.

nallath, 2017-04-12:
The empty material is something we should have a look at, same goes for the printer not working unless it has 2 cartrdiges in there. I don't think it will get a high priority though.

seantapscott, 2017-04-12:
Possible Workaround:
The first show-stopper I reached when trying to print with just one printcore was the Active Bed Leveling.  After it performed the bed leveling, it said the values for the bed leveling were too far outside of acceptable parameters, and aborted the print.  

I think that if the users changes Bed Leveling to off or manual, they may be able to proceed further, although I really like the Active Bed Leveling, myself.

Ghostkeeper, 2017-04-13:
I asked around about this.

The hardware team says that with the print core missing, the air flow in the head is messed up and we can't guarantee the same print quality.

The firmware team says that active levelling fails if the second print core is missing because the magnetic properties of the head are different and the z-offset calibration is obviously impossible. If you disable active levelling it could in theory work as long as the g-code never mentions T1 (and it's T1 that is missing). However, they plan to actively forbid this because of the reason that hardware mentions.

Cura should report the missing core properly. What is it reporting now, Nallath? I thought it was working fine since the new print monitor screen that I made.

fieldOfView, 2017-04-13:
Note that the UM3 & Cura could/should still support printing with one _material_ unloaded.

nallath, 2017-04-13:
I know what it reports if you have no material. It returns an empty GUID. Cura fails and uses whatever was active before (instead of having a special "empty" material).

seantapscott, 2017-04-17:
One other thing that I thing bears mentioning for this issue (again, trying to help) is that I can't find a way to make the skirt/brim print with only one material.  It always uses both materials, even if you specify only one.  Priming might also be an issue if one of the print cores has no material loaded. 

- Skirt/brim always prints using both materials. 
- Priming seems to always use both materials.

nallath, 2017-04-18:
It should only print brim with materials that are actually used in the print. Same goes for priming. Are you sure that your second extruder isn't used anywere?

seantapscott, 2017-04-18:
I think it was.  Printed yesterday and it didnt use second color, so
must've been a hidden setting the other times.

BagelOrb, 2017-07-23:
@nallath No it is always printing the brim with the Build Plate Extruder. If you really want a single extrusion print on the second extruder only you should set all Extruder settings to the second extruder.

It might be a nice feature to automatically set all extruder settings to the right extruder if all models are printed fully with the right extruder, though..

Ghostkeeper, 2017-08-04:
Worth mentioning that currently we see the bug that the front-end still thinks the first extruder is always used regardless of settings.

ChrisTerBeke, 2018-05-18:
Cura 3.3+ now support this (right click on an extruder, then disable it). This allows for one-at-a-time mode on UM3 and other dual extrusion machines, and also ignores all the settings of that extruder.

closed 2018-05-18 by ChrisTerBeke

seantapscott, 2018-05-19:
Sounds amazing.