Ultimaker/Cura #1673: Pause at height and redo layers

JPFrancoia, 2017-04-11:
Hi,

I'm using cura 2.5 beta (downloaded from the website), on Linux.
I have a problem with the pause at height plugin. When I choose to redo some layers (let's say 2 layers) after a pause, I can see in the Gcode that the layers a redone **before** the pause. It's not the expected behaviour.

[UM2O_POC_extraction_cartridge_broken.txt](https://github.com/Ultimaker/Cura/files/913227/UM2O_POC_extraction_cartridge_broken.txt)

I attached my gcode. I would like to redo layers 81 and 82 after the pause. But you can see in the Gcode that these layers are redone before the pause. After the pause, the printing resumes at layer 83.

Gcode unaltered directly outputed by Cura.

Is there a way to solve this issue ?

Ghostkeeper, 2017-04-13:
Indeed, this reproduces for me too (on master). Good find.

Ghostkeeper, 2017-04-13:
Duplicates https://github.com/nallath/PostProcessingPlugin/issues/41

closed 2017-04-13 by Ghostkeeper

JPFrancoia, 2017-04-13:
Yep, didn't know where to open the issue. Sorry :/