Ultimaker/Cura #1674: 2.5.1 Beta 2: Crash on Launch

CCS86, 2017-04-11:
Win10 x64

Splash screen comes up (but says 2.5.0 BETA, not 2.5.1 BETA 2). Right as the front end tries to load, it crashes.

[cura.log.txt](https://github.com/Ultimaker/Cura/files/913307/cura.log.txt)
[stderr.log.txt](https://github.com/Ultimaker/Cura/files/913310/stderr.log.txt)

fieldOfView, 2017-04-11:
> Splash screen comes up (but says 2.5.0 BETA, not 2.5.1 BETA 2).

The splashscreen is right. There is no 2.5.1 beta, there is a 2.5.0 beta and a 2.5.0 beta 2. Yes, the website is wrong.

CCS86, 2017-04-11:
Gotcha. the splashscreen doesn't say "BETA2" though.

fieldOfView, 2017-04-11:
I can't see anything that indicates a crash in the logs you provided.

CCS86, 2017-04-11:
It's a "..... stopped working" type Windows crash.

CCS86, 2017-04-11:
The BETA1 runs just fine for me. Let me know what other diag data I can provide. Can do a screen recording too.

awhiemstra, 2017-04-11:
You should get an APPCRASH report in your system logs. It might contain something useful, so posting that information may help.

CCS86, 2017-04-11:
[Report.wer.txt](https://github.com/Ultimaker/Cura/files/913638/Report.wer.txt)

awhiemstra, 2017-04-11:
Hmm I was hoping for some kind of backtrace... The only thing it says is this:

> Sig[3].Name=Fault Module Name
> Sig[3].Value=Qt5Qml.dll

Which is not very useful... I am also afraid that I do not see anything in the recent commits that would cause such a crash.

Ghostkeeper, 2017-04-13:
It also says:

> Sig[6].Name=Exception Code
Sig[6].Value=c0000005

Exception code c0000005 is for access violation ([source](http://stackoverflow.com/questions/17168982/exception-error-c0000005-in-vc)). So it could be a nullpointer or segfault type of thing. Looks like the bug is in Qt itself. But we didn't change the Qt version recently as far as I know.

ChrisTerBeke, 2018-03-02:
old beta. closing.

closed 2018-03-02 by ChrisTerBeke