Ultimaker/Cura #1675: PS_ON (M80) before M190,M104,M109

kr15uk, 2017-04-11:
Ladies and gents,

I need M80 (power on) command to be initiated **before** it kicks in heaters.
If I add M80 to my start-gcode, final gcode has M190,M104, M109 and only then goes M80...
Obviously no M80 = no 12v and no 12v = no power to the heaters...

Is there a chance to add M80 aka PS_ON tickbox in machine settings for the future releases?
Is there a workaround for the time being like editinh json file or so?!
Thanks.

I've attached 2x screenshots. 1st start-gcode in Cura, 2nd final gcode.

Thanks.

[image]

[image]

fieldOfView, 2017-04-11:
CuraEngine detects if your start gcode sets the temperatures. If it doesn't (like in your case), it adds them before the start gcode in the final gcode file.

Make sure you include M190 S{material_bed_temperature} and M104 S{material_print_temperature} in your start gcode (after the M80 line) and CuraEngine will not put them in the gcode before the start gcode.

BagelOrb, 2017-07-23:
Duplicate of https://github.com/Ultimaker/Cura/issues/1657

closed 2017-07-23 by BagelOrb