Ultimaker/Cura #1676: Add Custom Dual Extrusion machine

kr15uk, 2017-04-11:
Ladies and gents,

Would be nice to be able to add custom Dual Extrusion machine via Cura new machine wizard thing.
(Including nozzle offset and prime tower size, I find that mixing/y-splitter machines needs slightly larger footprint tower)

Currently doing it via custom json files but when tweaking start/end gcode I need to completely remove machine, kill Cura, edit json file, start up Cura and add machine once again otherwise Cura fails to start if any changes to json file is done without removing the machine.

Not urgent/crucial but would be nice to have in future releases.

Thanks.

fieldOfView, 2017-04-11:
See https://github.com/Ultimaker/Cura/pull/1521 which adds exactly that.

Note that you should not have to remove the machine to edit the extruder start/end gcode; closing cura, editing and relaunching should suffice. Prime tower size is already settable through the normal gui/settings.

Ghostkeeper, 2017-04-13:
Duplicate of https://github.com/Ultimaker/Cura/issues/1625

closed 2017-04-13 by Ghostkeeper