Ultimaker/Cura #1678: AtributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

dintid, 2017-04-12:
Using Cura. 2.5.0 Beta

Been using just fine for a bit and have used new materials. Today cura crashes with this error message when changing material or try to save changes to it.

> Version: 2.5.0-BETA
> Platform: Windows-10-10.0.14393
> Qt: 5.7.1
> PyQt: 5.7.1
> 
> Exception:
> Traceback (most recent call last):
>   File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 764, in setActiveMaterial
>   File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 828, in setActiveQuality
>   File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 944, in _replaceQualityOrQualityChangesInStack
> AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

Ghostkeeper, 2017-04-13:
The error seems to occur here: https://github.com/Ultimaker/Cura/blob/701570283227bbe717e3b6d7f121fcfa0d1d9990/cura/Settings/MachineManager.py#L923

It is called from one of these two: https://github.com/Ultimaker/Cura/blob/701570283227bbe717e3b6d7f121fcfa0d1d9990/cura/Settings/MachineManager.py#L807
So the quality or quality changes profile seems to be `None`.

Ghostkeeper, 2017-04-13:
I can't reproduce this yet though.

dintid, 2017-04-14:
Anything I can do to help sort this? What's odd is that I have created other materials without any issues.

Ghostkeeper, 2017-04-14:
Maybe you could save a project file while you have one of those materials that gives the bug active?
It's under File -> Save Project.

dintid, 2017-04-14:
I can't. As soon as I change to the material Cura crashes. It's only 1 material. I have an identical one, aside from the color, so it's a bit odd
[image]

I just noticed the profiles for the printer, where i use the material has gone missing, and I'm unable to create any - I can adjust the "active profile" though. Meaning I can change settings in main window.
[image]

Ghostkeeper, 2017-04-18:
Well, I'm going to need to know what printer and materials you're using then, and where to download them. By default, I don't see any DuetUME3D printer, or PrimaSelect material.

dintid, 2017-04-18:
It's just a custom printer I added, and added a material to that printer, which I copied from one of the ultimaker ones.
I havn't had time to create a new custom printer yet, which is what I planned to do next.

synthead, 2017-05-19:
I am experiencing this.  I'm running Cura 2.5.0 on Arch Linux.

Steps to reproduce:

1. Open Cura and observe prompt for a printer.
2. Select *Ultimaker 2 Extended+* and click the *Add Printer* button.
3. Click the yellow-highlighted drop-down selection on the top-right and click *Manage Profiles*.
4. Click *Printers* from the selection on the left, click *Ultimaker 2 Extended+*, and click the *Activate* button.
5. Click *Profiles* from the selection on the left.
6. Observe segmentation fault as detailed below.

Segmentation fault:
```
2017-05-18 18:07:20,905 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-18 18:07:20,906 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-18 18:07:20,906 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/Settings/MachineManager.py", line 1091, in getDefinitionByMachineId
2017-05-18 18:07:20,906 - CRITICAL - cura.CrashHandler.show [39]:     return containers[0].getBottom().getId()
2017-05-18 18:07:20,907 - CRITICAL - cura.CrashHandler.show [39]: AttributeError: 'NoneType' object has no attribute 'getId'
2017-05-18 18:07:22,000 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Ultimaker 2 Extended+, which doesn't exist.
2017-05-18 18:07:22,001 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Ultimaker 2 Extended+, which doesn't exist.
2017-05-18 18:07:22,004 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Ultimaker 2 Extended+, which doesn't exist.
2017-05-18 18:07:22,005 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-18 18:07:22,005 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-18 18:07:22,006 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/Settings/MachineManager.py", line 507, in activeMaterialNames
2017-05-18 18:07:22,006 - CRITICAL - cura.CrashHandler.show [39]:     if ExtruderManager.getInstance().getActiveGlobalAndExtruderStacks() is not None:
2017-05-18 18:07:22,007 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/Settings/ExtruderManager.py", line 418, in getActiveGlobalAndExtruderStacks
2017-05-18 18:07:22,007 - CRITICAL - cura.CrashHandler.show [39]:     result.extend(self.getActiveExtruderStacks())
2017-05-18 18:07:22,008 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/Settings/ExtruderManager.py", line 429, in getActiveExtruderStacks
2017-05-18 18:07:22,008 - CRITICAL - cura.CrashHandler.show [39]:     for extruder in sorted(self._extruder_trains[global_stack.getId()]):
2017-05-18 18:07:22,008 - CRITICAL - cura.CrashHandler.show [39]: KeyError: 'Ultimaker 2 Extended+'
2017-05-18 18:07:22,009 - WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [182]: Could not get metadata of container  because it was not found.
2017-05-18 18:07:22,009 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-18 18:07:22,010 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-18 18:07:22,010 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/UM/Settings/Models/InstanceContainersModel.py", line 60, in _update
2017-05-18 18:07:22,010 - CRITICAL - cura.CrashHandler.show [39]:     self._instance_containers = self._fetchInstanceContainers()
2017-05-18 18:07:22,010 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/Settings/UserProfilesModel.py", line 25, in _fetchInstanceContainers
2017-05-18 18:07:22,011 - CRITICAL - cura.CrashHandler.show [39]:     machine_definition = quality_manager.getParentMachineDefinition(global_container_stack.getBottom())
2017-05-18 18:07:22,011 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/QualityManager.py", line 251, in getParentMachineDefinition
2017-05-18 18:07:22,011 - CRITICAL - cura.CrashHandler.show [39]:     machine_entry = machine_definition.getMetaDataEntry("machine")
2017-05-18 18:07:22,011 - CRITICAL - cura.CrashHandler.show [39]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'
2017-05-18 18:07:22,011 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-18 18:07:22,011 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-18 18:07:22,011 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/UM/Settings/Models/InstanceContainersModel.py", line 60, in _update
2017-05-18 18:07:22,012 - CRITICAL - cura.CrashHandler.show [39]:     self._instance_containers = self._fetchInstanceContainers()
2017-05-18 18:07:22,012 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/Settings/UserProfilesModel.py", line 25, in _fetchInstanceContainers
2017-05-18 18:07:22,012 - CRITICAL - cura.CrashHandler.show [39]:     machine_definition = quality_manager.getParentMachineDefinition(global_container_stack.getBottom())
2017-05-18 18:07:22,012 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/QualityManager.py", line 251, in getParentMachineDefinition
[... cut ...]
2017-05-18 18:07:36,042 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.6/site-packages/cura/QualityManager.py", line 251, in getParentMachineDefinition
2017-05-18 18:07:36,042 - CRITICAL - cura.CrashHandler.show [39]:     machine_entry = machine_definition.getMetaDataEntry("machine")
2017-05-18 18:07:36,043 - CRITICAL - cura.CrashHandler.show [39]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'
Segmentation fault (core dumped)
```

Also, after selecting a printer, Cura does not appear to render a 3D representation of the printer like it does on OS X.  This is what I see:

[image]

Ghostkeeper, 2017-05-19:
Thanks for the very clear reproduction steps, synthead! They don't reproduce the problem on my end though. At step 3 it's not yellow-highlighted already. The rest of the steps can be performed but it's not producing the problem you describe.

It sounds like your Cura instance can't even find the UM2+ quality profiles. Apparently that's giving trouble. Either that, or adding the printer broke down somewhere halfway...

BagelOrb, 2017-07-23:
@dintid Are yous till experiencing this problem?

closed 2017-07-27 by nallath