Ultimaker/Cura #1684: Cura 2.4.0 Text Disappears

pearson, 2017-04-12:
This is an odd bug. I'm testing Cura 2.4.0 in a VMware VM running Windows 10 Pro 64 bit and, after a short amount of time, all text displayed by Cura either completely or partially vanishes. 

I suspect it might be due to the VM as I don't see this issue reported here at all. However, I have never seen a similar problem with other applications in these VMs.

[image]

awhiemstra, 2017-04-13:
This sounds like a problem with the graphics driver. We use OpenGL for most of the UI rendering. Since the passthrough drivers for most virtual machines are not the best, you get issues like this. There is not much we can do about that I'm afraid.

Ghostkeeper, 2017-04-13:
Seems that everything we render with "label" is rendered fine, but the stuff that is rendered with "text" is not.

This is something between OpenGL, your virtual machine and Qt.

closed 2017-04-13 by Ghostkeeper

fieldOfView, 2017-04-13:
> Seems that everything we render with "label" is rendered fine, but the stuff that is rendered with "text" is not.

Wut? We don't use Text nodes anywhere (except in OpenFilesIncludingProjectsDialog.qml, where we really should not because it results in the wrong font being used there - let me create a PR for that)

Ghostkeeper, 2017-04-14:
We are now:
https://github.com/Ultimaker/Uranium/commit/e8d7f72de2abf80a8abd3bb69080fa0ad55b4e77
https://github.com/Ultimaker/Uranium/commit/085ec47b7721acfce3e70af6d570b4ed44b70133
Searching for the Text element in QML yields me like 10 results at the moment.

We found that on some computers, things with "Label" get rendered as the system font. We then change the font face, size and colour, but the aliasing was still off. We figured that there were some rendering flags that we didn't change but didn't find how to change them. So we just switched to "Text".