Ultimaker/Cura #1687: In "Print Setup - Custom" the "Diameter" is not used from my material settings

gdachs, 2017-04-13:
I have made an entry for my own material. It has a diameter of 1.75mm. If I choose this material click on "Custom" the diameter 2.85 is shown. I have to enter 1.75. If I click on "Recommended" and again "Custom", then the diameter is again on 2.85.

Ghostkeeper, 2017-04-14:
I don't reproduce this.

However it is a known bug that the diameter of 1.75 that you enter in the material manager is not actually used when slicing and can be different from the diameter that you enter in the Custom settings pane.

gdachs, 2017-04-14:
I noticed now that it doesn't happen if I use for my printer the settings of Prusa i3 Mk2 and modify them to my needs. If I create a fresh custom entry for my printer, then it happens. Even that the diameter is no visible part of the printer settings.

Ghostkeeper, 2017-04-18:
All right, so these are my steps:
- Clear settings.
- Start Cura.
- Select Custom FDM Printer to add. Click away the settings action window.
- Go into the material manager.
- Duplicate PLA.
- Select the new PLA duplicate in the manager. Change the display name to BLA. Close the manager.
- Select the new BLA material.
- Under Custom, modify the material diameter to 1.75.
- Click Recommended.
- Click Custom again.

It now properly displays 1.75 under material diameter in the Custom screen. I still don't reproduce it.

Perhaps it has been fixed since 2.4? I did my test on Master.

gdachs, 2017-04-18:
Maybe there is another bug involved I noticed since then. I change the brand of material too, so that another group shows up. This group and the material disappears completely after restart. I will test again without changing the brand.

gdachs, 2017-04-18:
I noticed now that you do not the same like me. I change the material diameter not in Custom, I change it already in the material settings. The diameter is a setting of the material, not a setting of the current print.

Ghostkeeper, 2017-04-18:
Ah, so then my reproduce steps become:
- Clear settings.
- Start Cura.
- Select Custom FDM Printer to add. Click away the settings action window.
- Click Custom mode (rather than recommended).
- Go into the material manager.
- Duplicate PLA.
- Select the new PLA duplicate in the manager. Change the display name to BLA.
- Change the diameter of the material to 1.75.
- Close the material manager.
- Select the new BLA material.
- Click Recommended.
- Click Custom again.
- Go into the material manager to see the material diameter.

It still shows 1.75, which is expected.

gdachs, 2017-04-18:
Okay, so maybe this is fixed in master and I have to wait for the next stable version. 

Could you please be so kind and do the above steps and additionally change the brand and check whether this survives a restart?

Ghostkeeper, 2017-04-19:
Changing the brand has no effect.

After the restart, the material is gone though. This appears to be the bug described in https://github.com/Ultimaker/Cura/issues/1104.

gdachs, 2017-04-20:
The bug is still in cura 2.5.0. 

I made a new filament entry with Diameter 1.75:
[image]
and Default Printing Temperature 215 °C:
[image]
In Print Setup Custom the temperature shows up, but the diameter not:
[image]
The temperature is changing in the Custom Settings, if I choose another filament, but the diameter never.

Good news: the custom filament group survives a restart!

ex-nerd, 2017-04-25:
Same here on MacOS Cura (both 2.4 and now still in 2.5).   If I manually update the material diameter in the settings, it actually notes that it's different from default and offers the `↺` marker to "revert" it to the "default" value of 1.75.

I did some grepping through my settings and it seems like maybe this default value is being loaded from my printer definition.  In other words, this bug might be better renamed to `printer definition for material diameter overrides material definition (not the other way around, as it should)`.

trivalik, 2017-05-10:
I use the cura 2.5 beta2 on windows. I use a custom printer and could not find any diameter settings for machine. Not in Cura GUI nor in filessystem "C:\Users\<name>\AppData\Local\Cura\machine_instances".
I copied the existing ABS profile and changed the diameter to "1.75". I sliced once with the original ABS profile and then with mine. The saved gcode file is the same. It was also asking to keep my changed settings, I pressed always keep.

Short: change material type or filament in a duplicate of a generc material make no effect. just the temperature is used.
Maybe because the filament tab is called "Information"?

fieldOfView, 2017-05-10:
A number of changes were made for Cura 2.6. Please retest when the Cura 2.6 beta is launched

PS: if you are still using Cura 2.5 beta 2, please update to the final release of Cura 2.5

dagnall53, 2017-05-15:
I have also come across this as an issue, using 2.5. 
It would seem sensible to me that the material diameter in the materials files should overwrite anything in the configuration. That way, I can measure my material and set the diameter in the materials file, and if I chose that material, that exact diameter will be used. 
I discovered this as my transparent ABS is 1.66mm, everything else so far is 1.75. !! 
I would expect Ultimakers to read the diameter from the magic RF tags on Ultimaker materials , and do something similar...
Looking forward to 2.6!.. thanks

ex-nerd, 2017-05-27:
I can confirm that this is still happening in the 2.6 beta.  I've removed all `material_diameter` settings from the quality config files (by hand in a text editor), leaving only `<diameter>` settings within my materials configs themselves.

The Cura UI usually shows a diameter of 1.75, and when I switch back and forth between materials with different settings (e.g. 1.71 for some lower quality stuff) the UI continues to say 1.75.

If I manually set the diameter to 1.71 and then switch to a material that is configured to the same, Cura pops up with its usual "settings changed" dialog and says that it will be resetting the 1.71 value to the "default" of 1.75.

Note: It looks like someone has submitted #1890 (in reference to #1886) as a fix for this issue.

DDeGonge, 2017-06-21:
This seems to still be happening in 2.6 Offical. It displays 1.75mm as being used, however when it comes time to slice it does not use this value or the set hotend temperature. It always slices thinking it is using 2.85mm filament and at 200*C. I have tried using custom print profiles, default ones, setting the values manually, and having them set by the material profile. None seems to help.

Oddly enough, bed temperature is set correctly, even though this is set by the material profile as well.

BagelOrb, 2017-07-23:
Wasn't this bug solved in 2.6.2? Please try that release.

trivalik, 2017-07-31:
It Looks like fixed in 2.6.2. Is there no changelog? Because my Windows Installation Shows me just for 2.6.1 a changelog.

Ghostkeeper, 2017-08-04:
It wasn't fixed in 2.6.2. The only thing that changed in 2.6.2 is the CPU architecture target of the MacOS build. If the bug doesn't reproduce in 2.6.2 you must be doing something different than before.

When we updated the change log for 2.7 we added an entry for 2.6.2 as well that mentions this briefly.

trivalik, 2017-08-09:
I used the same sequence of steps to reproduce. I select 2 different self created filaments with different diameter. I used the last 2.6 beta so maybe in between was some change.

dagnall53, 2017-08-10:
I have tried 2.6 release version 2.6.2 with  two extruders, and two filaments, set at  1.7 and 1.6 and Cura now seems to recognise each filament individually. 
I think this is now closed.

gdachs, 2017-08-10:
I tried 2.6.2 and I can't reproduce the issue anymore.

closed 2017-08-10 by gdachs