Ultimaker/Cura #1689: Custom profiles are not showing up in the profiles list

juliusbangert, 2017-04-14:
When I create a new profile from the 'create profile from current settings' option under settings>profile; it seems to save it but invisibly.. If I ever select one of the 'protected profiles' I can never go back to my custom profile. You can see in the screen shots than on the right hand pull down I have a profile called M200, but this is neither selectable in the preferences or the pulldown itself.

The same applies to when you import a profile file.

[image]
[image]

Hope that all makes sense.
Cheers for the excellent software.

nallath, 2017-04-14:
Could you share: log file, project file & what version of Cura you are using?

nallath, 2017-04-14:
I've just tried it with the maylan m180 (when running it form source), and there it seems to work.

juliusbangert, 2017-04-14:
It's the recent update to 2.4. How do I share log file and project file?

nallath, 2017-04-14:
Host it somewhere so we can reach it. 

Another option would be to try the 2.5 BETA and see if that still has the issue. Do backup your local files before doing so though.

Ghostkeeper, 2017-04-18:
The log file can be found by going in Cura to Help -> Show Configuration Folder. The log should be there.

A project file can be created by goint into File -> Save Project. This saves the models on the build plate as well as the current machine, material, profiles and settings that you use, so that hopefully a slice is completely reproduced.

If you put the files in a .zip you can drop it in the comment box of this thread and Github will host it for you. If it's too big for Github, either trim the log file or host it on Mega instead.

mikosik, 2017-06-01:
I have very similar issue on cura 2.5.0 on Linux manjaro.
Preferences -> profile is completely empty even though I've just saved profile "perunlabs_next" by choosing settings -> profile -> "create profile from current settings/overrides".

[image]

logfile:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/1044041/cura.log.zip)

project file:
[cura_project_file.zip](https://github.com/Ultimaker/Cura/files/1044047/cura_project_file.zip)

Ghostkeeper, 2017-06-05:
> 2017-06-01 09:51:51,746 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Ultimaker 2+, which doesn't exist.

Seems to be that weird 404 error again... Still don't know how to reproduce that.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-23:
Yes, loading Mikosik's project reproduces this bug in 3.0.4.

Ghostkeeper, 2017-11-23:
I think this arose from defining a custom material type: PLA/PHA. This material type then has no profiles, but a quality profile was activated nonetheless. You can create a custom profile which then gets based off that quality type. The profile manager displays a list of profiles until restarting Cura. After restarting the list becomes empty.

Ghostkeeper, 2018-01-01:
Our project manager removed this item from our planning because it's older than 12 weeks. Still broken though even with the refactors we had for 3.1.

Ellecross, 2018-04-11:
Hi @juliusbangert and @mikosik! 

So it's been a year, :D but I'm happy to let you know that we checked your reported issues and found them fixed in 3.3 Beta. The custom profiles are now visible in the profile drop-down menu and in the profile manager window.  

This fix will also be in the final release of 3.3. 

Here's a screenshot with the custom profile from @mikosik's project file. 
[image]

If you find other issues related, let me know.

closed 2018-04-11 by Ellecross