Ultimaker/Cura #1691: Cura and windows 10 creator update

Spamouille, 2017-04-15:
I have to rename the arduino directory to arduino.bak to be able to start cura 2.4/2.5 with windows 10 creator update

Spamouille, 2017-04-15:
I tried to install arduino driver manually "The hash for the file is not present in the specified catalog file"

Ghostkeeper, 2017-04-18:
I'm tracking an issue for this. It sounds quite serious!

Our system testing team is asking around if anyone got the creator's update yet. One of our QA testers is installing the update on his home computer tonight (because he doesn't want to have this problem prevent his testing at work).

Ghostkeeper, 2017-04-18:
What happens when you start Cura? Does it create a log in `C:\Users\<Spamouille>\AppData\Local\cura`?

Spamouille, 2017-04-18:
i don't have the cura.log of the moment the issue occured but i remember the last entry was :
this one ->DEBUG - UM.Controller.setActiveView [81]: Setting active view to LayerView

i don't have the issue anymore... even if i rename the arduino directory to the normal name.

randyxxl, 2017-04-18:
I can't reproduce the problem. I installed 2 new versions, ran all versions of Cura available on my machine, installed the Creator Update and opened all versions of Cura again. Before and after the Creator update Cura runs without any problem and without any unexpected error in the logs.

Tested versions:
- Cura 2.5.0 Beta (internal release of 2017-03-22)
- Cura 2.5.0 (today's Release Candidate)
- Cura 2.6.0 (today's Master Build)
Each version was installed using the default settings. Only the firewall bypass rule was rejected by me.

The 2.5.0 RC reused the settings of the earlier installs.
The 2.6.0 can be considered a clean install as of this version each version (2.6, 2.7 etc.) has it's own config folder and we haven't provided a upgrade scripts yet.

So the Creator Update seems not to be the problem. 

Please check cura.log and stderr.log again if the problem returns.

closed 2017-04-18 by Spamouille