Ultimaker/Cura #1692: 2.5.0 Beta 1: Visualization "skinny"

CCS86, 2017-04-15:
The Cura visualizer (which is awesome overall), is showing some skinny segments, that should be full width:

[image]

Repetier:
[image]

Gcode:
[UMO_HandleSpacerp.gcode.txt](https://github.com/Ultimaker/Cura/files/923661/UMO_HandleSpacerp.gcode.txt)

nallath, 2017-04-16:
The issue is in our current sprint. @BagelOrb also found it ;)

BagelOrb, 2017-04-17:
I'm afraid it isn't. There seems to be something wrong with the overlap compensation algorithms. It seems to be compensating a segment of the walls where it should have been compensating the segment next to it.

Where the outer wall makes a sharp corner on the bottom right, the small piece that's left at normal extrusion width should have been reduced because it is overlapping with the bottom part of the outline. Instead the segment following it is reduced.

This should be easily reproducible with a small wedge.

In case it's not: please share your project file. Sharing the gcode is actually not of much use; it only shows *that* something is wrong, but it doesn't let us reproduce *what* goes wrong.

Appesteijn, 2017-04-25:
This model: http://www.thingiverse.com/thing:1784375 has the under extrusion at layer 1

( https://ultimaker.com/en/community/45415-under-extrusion-on-some-shells-wall-lines )

BagelOrb, 2017-04-25:
> This model: http://www.thingiverse.com/thing:1784375 has the under
> extrusion at layer 1
>
> ( https://ultimaker.com/en/community/45415-under-
> extrusion-on-some-shells-wall-lines )

( CURA-3729 )

ianpaschal, 2018-02-27:
We won't be doing any more work on versions below 2.7 so this is a `Won't Fix`.

closed 2018-02-27 by ianpaschal

BagelOrb, 2018-02-27:
You mean: this bug has been fixed since that version, so it can be closed.

Bugs aren't logged to be solved in that specific version; they are logged
on the version they occur in to be fixed in some newer release.

ianpaschal, 2018-02-27:
Yes, I know how bugs work haha. But some people don't, I guess, and expect fixes for their legacy version rather than upgrading to latest.

Seeing as there was 700-750 open issues for the last few weeks, I've been trying to get a handle on it. If it's an issue reported in version 2.7-3.2.1, I check if it's fixed. If it's below that, I just close and say we won't be working on 2.6 and older.

I know I'm sort of hacking with a blunt knife here but there's no way there's actually still 640 features under consideration and bugs which are either in progress or being considered.

BagelOrb, 2018-02-27:
If you're closing it because it's a too old issue then you should say
*that* rather than saying that we won't work on that version. Not working
on old versions has never been a reason to close an issue and you say it's
not even your reason, so don't say it. Say the real reason instead.

BagelOrb, 2018-02-27:
The real reason being that there are too many issues open and the issue is
too old.

On 27 Feb 2018 11:27 a.m., "Tim Kuipers" <t.kuipers@ultimaker.com> wrote:

If you're closing it because it's a too old issue then you should say
*that* rather than saying that we won't work on that version. Not working
on old versions has never been a reason to close an issue and you say it's
not even your reason, so don't say it. Say the real reason instead.

ianpaschal, 2018-02-27:
> If you're closing it because it's a too old issue then you should say *that* rather than saying that we won't work on that version. Not working on old versions has never been a reason to close an issue.

The ticket age and the version age are obviously closely linked, and my point was that I care about what version it relates to and that _version_ is too old to worry about. Closing issues because they relate to unsupported versions is absolutely valid.

Anyway, I just double checked and I marked it won't fix because that's what's in JIRA for this ticket.

BagelOrb, 2018-02-27:
Actually this bug persists to this day in all versions since it was first
logged.

The title is a bit deceptious though.