Ultimaker/Cura #1693: custom post processing scripts are not saved to a project file

UltiArjan, 2017-04-15:
Cura 2.5 Beta2. 
Win7

I noticed custom post processing scripts are not saved to a project file, which is pretty difficult if you use a printer setup depending on it. I guess it's not a bug but just not implemented jet? In that case please consider it.....

In the case you open the project file on a machine that does not have the post processor installed, it should at least give a warning...

nallath, 2017-04-16:
It's not implemented by "design". As far as I see, no machine should require post-processing plugin by default. 
It would also take quite a bit of effort to get it working right and as no Ultimaker machine needs it, I don't think anyone from UM is going to build it. Pull requests regarding this are obviously always accepted.

Ghostkeeper, 2017-04-18:
For reproducibility of post-processing related bugs, I think it'd be useful to save this as well. However, for security purposes, perhaps we should only save the name of the script and the settings that were used.

Ghostkeeper, 2017-08-08:
This was just rejected by our project manager for being inactive for 3 months.

nallath, 2019-07-16:
This has been implemented some time ago.

closed 2019-07-16 by nallath