Ultimaker/Cura #1695: List/Label Objects to Allow Cancelling Split Part Jobs

xraymancouk, 2017-04-16:
_This may already be a repeat of another question but I struggled to find anything similar in the list._

I often print multiples of the same item every so often the techABS I've printed with has lifted on one part which then ruins the entire print by either drifting into the other print or causing buildup on the nozzle. If I could just tell the print to kill that one part and leave the others it would definitely make life better.

Would this be a useful feature for others?

This follows on from a discussion on the cel-robox forum. I asked if it was possible to cancel one part of a multiple part print to prevent failing the entire print due to one part lifting.

'@saiken' - suggested contacting the Cura team directly and asking "...is have a list of objects in the header of the gcode, and then delimit the object being printed on each layer before its gcode. The firmware could keep a list of objects in memory, and if one is marked bad then it gets ignored on subsequent layers."

nallath, 2017-04-16:
As far as I can see it, this would be quite hard to implement, let alone get right / understandable for the user.

Ghostkeeper, 2017-04-18:
It also messes quite a lot with the idea of g-code just being a set of instructions for the printer to follow. I think a better idea is to allow temporarily disabling parts of the print in Cura, so that the user can more easily re-start his print of a full build plate when some parts succeeded and some parts failed. This needs to have a simple and well-hidden workflow though.

BagelOrb, 2017-07-21:
This needs a rework of the firmware and the file format known as gcode.

This is most definitely not going to happen for your current machine.

closed 2017-07-21 by BagelOrb

Sophist-UK, 2020-05-23:
Octoprint now has a Cancel Object plugin that can do just that **PROVIDING** that the Gcode has appropriate comment markers to show when the slicer finishes a layer on one object and starts a layer on another object.

Since this has been closed several years ago, I have added a new issue for it at #7812.

Sophist-UK, 2020-05-23:
See also #7315 and #7014.