Ultimaker/Cura #1696: rounding errors in Cura

mypeopoly, 2017-04-16:
Hi,
I was talking with Tim Kuipers on Facebook and he informed me that Cura rounds the calculation to the integer micron in a discussion about using Cura for SLA printing.  He is concerned the accumulated errors may be up to 10 microns. I just want to learn more about the inner working of Cura in this area.  What is the best place to dig in and learn more

Ghostkeeper, 2017-04-18:
Start with the `utils/LinearAlg2D.cpp` file. The accumulated errors he mentions are the result of CuraEngine working with integer math. The integer math is in there: It has to do some vector math over there, and multiplies distances with small-ish direction vectors. This is the result of most of CuraEngine's inaccuracies, as far as I know.

mypeopoly, 2017-04-19:
@Ghostkeeper  I found those files in the Curaengine section. How do I go about testing them? Do I complie the Curaengine and replace the one in my Cura 2.4?
I have some programming experience from math course works but not much with C++

Ghostkeeper, 2017-04-19:
A good programmer would write unit tests to test them. We already have some tests in the `tests/utils/PolygonUtilsTest.cpp` file and such that should serve as an example. Basically, you'd write a function that has assertions in it. You test the assertion using `CPPUNIT_ASSERT_MESSAGE` or other such functions that the CppUnit library provides. Then you register those functions in the accompanying header file. If you then compile CuraEngine (including tests) using CMake, it'll produce additional executables that will yield an `output.xml` file when ran, with the results of the test.

A less experienced programmer would just look for the main function in `src/main.cpp`, which is the entry point of the program. There you can just call the Engine's functions manually, and print the results into `std::cout`. Put a `return 0;` after your calls so that you don't run the rest of the engine every time.

If you make tests in the first way we can consider merging them and we'll run those tests regularly to confirm that nothing breaks. If you test things in the second way we can't, obviously.

mypeopoly, 2017-04-27:
@Ghostkeeper 
Thanks for such detailed reply. I am very inexperienced when it comes to cpp so it would take some times to dig into to that.  Perhaps let me share with you where I became interested in this topic and like to find out to know if I am attacking the right problem.

As mentioned, we are using Cura to generate toolpaths for laser SLA printer.  One thing we noticed is that in some situations, Cura may skip some of the small features of an object. We were wondering if this is due to rounding errors in which the small features are rounded to zero.

An example is here:
here is an STL for a ring with support built in:
https://drive.google.com/file/d/0Bzke6lBHG_z5NEFJYzdoVlZIU3M/view?usp=sharing

The tip of support attached to the top of the rings are very small:
Cura Normal View
[image]
Cura Layer View
[image]

Is rounding the source of this issue?  btw, the laser spot size is 0.067mm and we use that as the nozzle size in Cura, as well as line width.

BagelOrb, 2017-04-30:
Try Cura 2.5! It has a way better layer view (if your graphics card supports it).

How thick are those thin round pieces at the top of the support?
Cura will only print those if they are about 0.1 mm wide approximately, due to your line width.

mypeopoly, 2017-05-03:
I did. 2.5 is much faster but still showing the same thing.  I even 
change the nozzle/line width down to 0.01mm and it still happens.

Ghostkeeper, 2017-05-04:
I tried this with the Peopoly definition in Cura 2.5 and it reproduces. Indeed, it appears as if the lines would easily fit in the vertical piece. At the last layer that is printed it still has 3 walls, even. X-Ray view shows no mistakes either.

It's not a problem with the PolygonProximityLinker. I disabled Compensate Wall Overlaps, and it still happens. You guys probably want to disable Compensate Wall Overlaps in your printer definition as well, since it depends on the material flowing to the correct places with overextrusion.

To debug this, I would start looking in FffPolygonGenerator to see if the correct number of walls are generated. You can debug polygons by using the SVG utility class to write a specific layer of polygons to an SVG file.

mypeopoly, 2017-05-15:
With the help of Moai user Paul Y, we found this section of the code:
[image]

The code "normal (ap, 1000),ap)/1000" is where CuraEngine scale up to take advantage of integer and vector math?  If so, what would happen if we increase 1000 to 10000. Would that increase the precision?  And would that significantly increase system resource required to run Cura? 
@Ghostkeeper @BagelOrb

BagelOrb, 2017-05-15:
That function is not really used that much. I don't think changing it to 10000 would change much in either the result or the time efficiency.

I think what's bothering you is slicer.cpp:773
That piece of code removes polygons which are too small.

mypeopoly, 2017-05-16:
@BagelOrb 
thanks for the quick reply. I will check into that as well.  

As with the code "normal (ap, 1000),ap)/1000" , could you share what it does? We also found some artifacts when printing and we wondered if it is related to rounding.

Ghostkeeper, 2017-05-17:
A true normal vector should have length 1. That doesn't work with integer math: There would only be 4 possibilities to have a vector of length 1 when the coordinates of that vector need to be round integers. Luckily, the `normal` function allows you to specify a length of your normal vector so you can get a bit more resolution out of it.

We then perform the dot product, and reduce the dot product back to what it would've been if we wouldn't have multiplied the normal vector.

cncpadawan, 2017-05-20:
Hello all!  I'm helping @mypeopoly with making these changes and seeing if they will work, but have been repeatedly stuck with trying to build in Windows.  I have VS2015 installed, as well as MinGW, and have been trying to follow the snippets of instructions on the github for all of the projects.  I was able to build protobuf, but had to use Google's repository for it.  Right now, I am trying to build libArcus, and it keeps failing.  First issue was that the install of protobuf didn't include all of the required header files, so I copied them over.  

Here is what I am getting when I am trying to build libArcus:
C:\Code\Cura\libArcus\cmake\build>cmake -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release -DProtobuf_LIBRARIES="c:\Protobuf\lib\libprotobuf.lib" -DProtobuf_INCLUDE_DIR="C:\Protobuf\include" ../..
-- The C compiler identification is MSVC 19.0.23918.0
-- The CXX compiler identification is MSVC 19.0.23918.0
-- Check for working C compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe
-- Check for working C compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe
-- Check for working CXX compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Protobuf: c:\Protobuf\lib\libprotobuf.lib (found suitable version "3.3.0", minimum required is "3.0.0")
-- Found PythonInterp: C:/Python36/python.exe (found suitable version "3.6", minimum required is "3.4.0")
-- Found PythonLibs: C:/Python36/libs/python36.lib (found suitable version "3.6.0", minimum required is "3.4.0")
-- Found SIP version: 4.19.2
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Code/Cura/libArcus/cmake/build

C:\Code\Cura\libArcus\cmake\build>nmake

Microsoft (R) Program Maintenance Utility Version 14.00.23918.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Scanning dependencies of target Arcus
[  4%] Building CXX object CMakeFiles/Arcus.dir/src/Socket.cpp.obj
Socket.cpp
c:\code\cura\libarcus\src\Error.h(103): warning C4251: 'Arcus::Error::_error_message': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'Arcus::Error'
D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\xstring(2628): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\code\cura\libarcus\src\Socket.h(148): warning C4251: 'Arcus::Socket::d': class 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'Arcus::Socket'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
c:\code\cura\libarcus\src\Socket.h(148): note: see declaration of 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
C:\Protobuf\include\google/protobuf/io/coded_stream.h(864): warning C4800: 'google::protobuf::internal::AtomicWord': forcing value to bool 'true' or 'false' (performance warning)
C:\Protobuf\include\google/protobuf/io/coded_stream.h(868): warning C4800: 'google::protobuf::internal::Atomic64': forcing value to bool 'true' or 'false' (performance warning)
[  9%] Building CXX object CMakeFiles/Arcus.dir/src/SocketListener.cpp.obj
SocketListener.cpp
c:\code\cura\libarcus\src\Error.h(103): warning C4251: 'Arcus::Error::_error_message': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'Arcus::Error'
D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\xstring(2628): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\code\cura\libarcus\src\Socket.h(148): warning C4251: 'Arcus::Socket::d': class 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'Arcus::Socket'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
c:\code\cura\libarcus\src\Socket.h(148): note: see declaration of 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
[ 13%] Building CXX object CMakeFiles/Arcus.dir/src/MessageTypeStore.cpp.obj
MessageTypeStore.cpp
C:\Protobuf\include\google/protobuf/io/coded_stream.h(864): warning C4800: 'google::protobuf::internal::AtomicWord': forcing value to bool 'true' or 'false' (performance warning)
C:\Protobuf\include\google/protobuf/io/coded_stream.h(868): warning C4800: 'google::protobuf::internal::Atomic64': forcing value to bool 'true' or 'false' (performance warning)
C:\Code\Cura\libArcus\src\MessageTypeStore.cpp(40): warning C4267: 'initializing': conversion from 'size_t' to 'uint32_t', possible loss of data
[ 18%] Building CXX object CMakeFiles/Arcus.dir/src/PlatformSocket.cpp.obj
PlatformSocket.cpp
C:\Code\Cura\libArcus\src\PlatformSocket.cpp(86): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
[... cut ...]
NMAKE : fatal error U1077: '"D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.

C:\Code\Cura\libArcus\cmake\build>

So, I am stuck here right now.  I have asked in the IRC group if there was a walk-through on how to get through all of the git projects and/or use the build scripts (those projects have failed as well), but didn't really get a response.  I am using the 2.5 tag in all of the repositories where that tag is applicable.  I am sure I am missing something simple, and would appreciate any help in  being able to get over this hurdle you can offer!

cncpadawan, 2017-05-20:
Also, just for reference, cura-build-environment is blowing up building OpenSSL.  When I re-run nmake after it errors, this is what I have for output:
C:\Code\Cura\libArcus\cmake\build>cmake -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release -DProtobuf_LIBRARIES="c:\Protobuf\lib\libprotobuf.lib" -DProtobuf_INCLUDE_DIR="C:\Protobuf\include" ../..
-- The C compiler identification is MSVC 19.0.23918.0
-- The CXX compiler identification is MSVC 19.0.23918.0
-- Check for working C compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe
-- Check for working C compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe
-- Check for working CXX compiler: D:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/amd64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Protobuf: c:\Protobuf\lib\libprotobuf.lib (found suitable version "3.3.0", minimum required is "3.0.0")
-- Found PythonInterp: C:/Python36/python.exe (found suitable version "3.6", minimum required is "3.4.0")
-- Found PythonLibs: C:/Python36/libs/python36.lib (found suitable version "3.6.0", minimum required is "3.4.0")
-- Found SIP version: 4.19.2
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Code/Cura/libArcus/cmake/build

C:\Code\Cura\libArcus\cmake\build>nmake

Microsoft (R) Program Maintenance Utility Version 14.00.23918.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Scanning dependencies of target Arcus
[  4%] Building CXX object CMakeFiles/Arcus.dir/src/Socket.cpp.obj
Socket.cpp
c:\code\cura\libarcus\src\Error.h(103): warning C4251: 'Arcus::Error::_error_message': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'Arcus::Error'
D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\xstring(2628): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\code\cura\libarcus\src\Socket.h(148): warning C4251: 'Arcus::Socket::d': class 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'Arcus::Socket'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
c:\code\cura\libarcus\src\Socket.h(148): note: see declaration of 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
C:\Protobuf\include\google/protobuf/io/coded_stream.h(864): warning C4800: 'google::protobuf::internal::AtomicWord': forcing value to bool 'true' or 'false' (performance warning)
C:\Protobuf\include\google/protobuf/io/coded_stream.h(868): warning C4800: 'google::protobuf::internal::Atomic64': forcing value to bool 'true' or 'false' (performance warning)
[  9%] Building CXX object CMakeFiles/Arcus.dir/src/SocketListener.cpp.obj
SocketListener.cpp
c:\code\cura\libarcus\src\Error.h(103): warning C4251: 'Arcus::Error::_error_message': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'Arcus::Error'
D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\xstring(2628): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\code\cura\libarcus\src\Socket.h(148): warning C4251: 'Arcus::Socket::d': class 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'Arcus::Socket'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
c:\code\cura\libarcus\src\Socket.h(148): note: see declaration of 'std::unique_ptr<Arcus::Socket::Private,std::default_delete<_Ty>>'
        with
        [
            _Ty=Arcus::Socket::Private
        ]
[ 13%] Building CXX object CMakeFiles/Arcus.dir/src/MessageTypeStore.cpp.obj
MessageTypeStore.cpp
C:\Protobuf\include\google/protobuf/io/coded_stream.h(864): warning C4800: 'google::protobuf::internal::AtomicWord': forcing value to bool 'true' or 'false' (performance warning)
C:\Protobuf\include\google/protobuf/io/coded_stream.h(868): warning C4800: 'google::protobuf::internal::Atomic64': forcing value to bool 'true' or 'false' (performance warning)
C:\Code\Cura\libArcus\src\MessageTypeStore.cpp(40): warning C4267: 'initializing': conversion from 'size_t' to 'uint32_t', possible loss of data
[ 18%] Building CXX object CMakeFiles/Arcus.dir/src/PlatformSocket.cpp.obj
PlatformSocket.cpp
C:\Code\Cura\libArcus\src\PlatformSocket.cpp(86): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
C:\Code\Cura\libArcus\src\PlatformSocket.cpp(112): warning C4244: 'initializing': conversion from 'SOCKET' to 'int', possible loss of data
C:\Code\Cura\libArcus\src\PlatformSocket.cpp(193): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
C:\Code\Cura\libArcus\src\PlatformSocket.cpp(236): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
[ 22%] Building CXX object CMakeFiles/Arcus.dir/src/Error.cpp.obj
Error.cpp
[... cut ...] yInputStream@123@@Z)
Arcus.dll : fatal error LNK1120: 1 unresolved externals
LINK failed. with 1120
NMAKE : fatal error U1077: '"C:\Program Files\CMake\bin\cmake.exe"' : return code '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"D:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\nmake.exe"' : return code '0x2'
Stop.

C:\Code\Cura\libArcus\cmake\build>

cncpadawan, 2017-05-21:
Hello,

Some updates.

I am now able to build CuraEngine in Windows!  Is Python support for libArcus required?  The GitHub page indicates you have to use Visual Studio if you want that support, but doesn't say it is required so it is a bit unclear.

There seems to be a problem with the cura-build-environment project.  It hangs when building sip, it seems to get into an infinite loop doing so (spawns over 8k processes).  I had that issue before building from source.  I solved it by compiling it using VC tools instead of MinGW if I recall correctly.

Ghostkeeper, 2017-05-21:
Hey Padawan. I'd like to say that when you want a response on the IRC channels (#ultimaker, #cura or #cura-dev) it's probably best to mention the user name of one of the devs, like me. We're usually around, but busy working and IRC just falls on the background (or another workspace). If you mention my nickname, except during lunch, I'll get a notification and should respond right away.

Secondly, Cura was made to be built with MinGW rather than MSVC. The only exception is that CPython requires their libraries to be built with the same compiler as CPython itself, which is MSVC. So the Python libraries we provide must be built with MSVC: libArcus, and consequently Protobuf as well. However CuraEngine also requires libArcus and Protobuf, so you'd compile them twice. We've had some pull requests of people building CuraEngine with MSVC as well, and reportedly they've had quite some success, but we don't actively develop with MSVC in mind unless it be for libArcus.

If you want to use Cura's front end, Python support of libArcus is required. This is because Cura's front-end is made in Python and can't talk to CuraEngine without libArcus. If you don't want to use the front-end, you can disable libArcus support when compiling CuraEngine and you won't need to have libArcus at all. The downside is that you can then only call CuraEngine from the command line.

I've never heard of that problem with Sip before! It's a bit out of our reach though, Sip being upstream. Perhaps the folks at Riverbank can help out. Or if you do find a solution you can post it here (or tell that to Riverbank).

awhiemstra, 2017-05-23:
@cncpadawan The OpenSSL build fails because it is trying to use MSYS's Perl, whereas you need an actual Windows version of Perl for OpenSSL to be able to build. Our build machines use ActivePerl for that, you can get it here: https://www.activestate.com/activeperl . Once you have it installed, make sure activeperl is used instead of msys perl.

cncpadawan, 2017-05-27:
@awhiemstra - thanks for the suggestion!  I was using Strawberry Perl, and, well, I installed ActivePerl and things seemed to have snapped out of it a little bit!

I am now successfully able to build cura-build-environment!  I have set my path to use that binary folder, and also added the lib folder to my path (not sure that is necessary, but...).

I am trying to now build in cura-build... but I get an error get_filename_component unknown component CACHE when I try to use cmake.  I have created a PFX certificate (self-signed) and specified the password... so that is the ONLY error I get there.  Any thoughts to what I may be missing there?

Here is the output and string I'm using for cmake: 
---
C:\Code\Cura\cura-build\build>cmake -G "MinGW Makefiles" -DBUILD_64BIT:BOOL=ON -DWINDOWS_IDENTITIY_PFX_FILE=..\signing.pfx -DWINDOWS_IDENTITIY_PFX_PASSWORD=<MY_PASSWORD_OMITTED> ..
-- The C compiler identification is GNU 5.1.0
-- The CXX compiler identification is GNU 5.1.0
-- Check for working C compiler: C:/TDM-GCC-64/bin/gcc.exe
-- Check for working C compiler: C:/TDM-GCC-64/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/TDM-GCC-64/bin/g++.exe
-- Check for working CXX compiler: C:/TDM-GCC-64/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:23 (get_filename_component):
  get_filename_component unknown component CACHE

-- Found Protobuf: C:/CuraBuild/lib/libprotobuf.lib (found suitable version "3.0.0", minimum required is "3.0.0")
-- Found PythonInterp: C:/CuraBuild/bin/python.exe (found suitable version "3.5.2", minimum required is "3.5.0")
-- Found PyQt: C:/CuraBuild/lib/site-packages/PyQt5 (found suitable version "5.7.1", minimum required is "5.6")
-- Found SciPy: C:/CuraBuild/lib/site-packages/scipy (found suitable version "0.19.0", minimum required is "0.17")
-- Found cx_Freeze: C:/CuraBuild/lib/site-packages/cx_Freeze (found version "5.0")
-- Found Signtool: C:/Program Files (x86)/Windows Kits/10/bin/x64/signtool.exe
-- Signing executables with: ..\signing.pfx
-- Configuring incomplete, errors occurred!
See also "C:/Code/Cura/cura-build/build/CMakeFiles/CMakeOutput.log".
---

There was nothing in the output log that I recognized as causing the failure.

Finally, I DID manage to build CuraEngine.exe going through the build process manually... libArcus failed building under MSVC so the Python component didn't get built.  In any case, CuraEngine works until I use a custom profile that I received from @mypeopoly.  Then CuraEngine throws an unknown software exception.  I am not sure that I am going to be able to attach to it using gdb to try and find out what is up there, do you have suggestions how you would trace something like that down?  It is something in that build process I followed - the 2.5 CuraEngine works fine with the profile.  I also have a profile for an he280 and it works with either my compiled CuraEngine or the default one, so... something in the profile is causing it to die on me.  I made no changes to the code before building, just to rule out anything I may have done there.  Thanks @Ghostkeeper for your feedback the other day in IRC, it helped me make some progress here I think!

In any case, I am hoping the cura-build / cura-build-environment will build things more successfully.

Ghostkeeper, 2017-05-29:
Relevant line in the CMakeLists.txt file: https://github.com/Ultimaker/cura-build/blob/master/CMakeLists.txt#L23

The CACHE component was already present in CMake 2.8.12 (which is the required version at the top of the file): https://cmake.org/cmake/help/v2.8.12/cmake.html#command:get_filename_component

Perhaps it's thinking that `FORCE` is a parameter for the `CACHE` keyword? Anyway, I can't seem to find any documentation of the use of `FORCE` in the `get_filename_component` function, not even in [the most recent version](https://cmake.org/cmake/help/v3.8/command/get_filename_component.html#command:get_filename_component) of CMake. It seems to be a parameter of the `set` function, but I don't know if that actually generalises to `get_filename_component`.

cncpadawan, 2017-05-29:
Hello @Ghostkeeper,

Thanks for your reply!  I did get it figured out with @ahiemstra's help... I was not specifying the build environment path.  The error wasn't very clear there...

I am still having crashes with the custom profile I have though... the build environment once built doesn't seem to build libArcus using MSVC (cmake by default wants to use MSVC and won't find the library).  I can make it all build using mingw, and it runs with default profiles from what I can tell, but it crashes with the custom profile (CuraEngine crashes more specifically).  The odd thing, running that profile with the installed version works just fine, ONLY the CuraEngine I build (using the build environment, or not) crashes.

Any suggestions on how to track down crashes such as these?  I am not sure if it is related to libArcus when other profiles work ok, but I don't know why this profile breaks on the CuraEngine I build but not the downloaded one.

Ghostkeeper, 2017-05-29:
If CuraEngine crashes, in general it's advisable to build CuraEngine with `CMAKE_BUILD_TYPE` set to `Debug` so that debug symbols are produced. Then run Cura with `python cura_app.py --external-backend` from a command line. Load a model in Cura and select your special profile or whatever causes it to crash. Then run CuraEngine separately via gdb using `gdb --args CuraEngine.exe connect -p 49674` (or whatever port your Cura command line is saying it's trying to connect to at the moment). In GDB, type `run` to start the engine. When it crashes, you can type `bt` in GDB to see a backtrace.

Ghostkeeper, 2017-05-29:
It also helps to find out which settings in your profile cause the engine to crash. You can just do this by resetting some of the values to the default in the Cura interface.

cncpadawan, 2017-05-29:
Hello @Ghostkeeper,

Thanks for the info!!!

May have figured it out, at least at a high level...  I have a SeeMeCNC H2 and was using the custom profile for an SLA, and it seemed to be causing the issue.  I added a new printer (SLA, it seems the build script is building out of master) and now... it isn't crashing.  I think I'm going to remove Cura in general from my system, then start fresh, maybe it is a corrupt profile somewhere along the line.

Is there a parameter I can specify to identify the specific tag I may want to use for cura-build that you are aware of?  Also... it seems to download from the repositories every time... is there an easy way (parameter or otherwise) to get it to stop doing that so I can make some code changes for testing?  I plan on digging into that tonight so I haven't looked yet, but if you know offhand, that would be awesome!

THANK YOU very much for your help!

cncpadawan, 2017-05-29:
It seems that I still have the crash issue...  but it is odd.  Using the code out of master, if I select the Moai profile that has been added for 2.6 beta and use my custom profile, it doesn't crash.  If I use a different printer profile (SeeMeCNC H2, Robo 3D as a couple examples I tried), or even create a custom FDM printer in the wizard, as long as I use that slicing profile it crashes.  

So, I wanted to try debugging it.  I ran cura.exe --external-backend and it waits for the CuraEngine as you have mentioned.  I used a little bit different syntax to fire off gdb (gdb --args CuraEngine connect 127.0.0.1:49674), after verifying the port it is after in the log.  When needed, I run it and I get a lot of warnings about unregistered settings, then at the end it produces 2 errors:

[ERROR] Couldn't find any extruder trains!
[ERROR] Couldn't load extruder.def.json for extruder 0

If I try running again, then CuraEngine displays the help info and sits there.  If I delete the model, it will stop executing.

Am I missing some command line options for CuraEngine to make it happy for this?  I apologize, I haven't run this via command line before.

awhiemstra, 2017-05-29:
Yes, you are missing the main printer definition, -j path/to/fdmprinter.def.json . 

As for building a specific branch, you can set the TAG_OR_BRANCH cmake option to specify what branch to build.

Assuming you keep the build directory the same, CMake should only do a git pull on each of the downloaded repositories. However, you may need to checkout a specific branch first. By default we build origin/master which is reset on every pull.

cncpadawan, 2017-05-30:
@awhiemstra, thank you for the info!  I am curious though.. should I still have to specify the printer definition if it is connecting to the front end?  The front end doesn't pass that info?  Will give specifying it a shot in a bit tonight. :)

cncpadawan, 2017-05-30:
This is what I get using gdb:
File: C:\Code\Cura\cura-build\build\CuraEngine-prefix\src\CuraEngine\src\pathPlanning\NozzleTempInsert.cpp, Line 16

Expression: temperature != 0 && temperature != -1 && "Temperature command must be set!"

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
[Thread 26892.0x6e9c exited with code 3]
[Thread 26892.0x6d64 exited with code 3]
[Thread 26892.0x146c exited with code 3]
[Thread 26892.0x27c8 exited with code 3]
[Thread 26892.0x80bc exited with code 3]
[Thread 26892.0x2c94 exited with code 3]
[Thread 26892.0x3880 exited with code 3]
[Thread 26892.0x8138 exited with code 3]
[Thread 26892.0x3a5c exited with code 3]
[Thread 26892.0x458c exited with code 3]
[Thread 26892.0x1954 exited with code 3]
[Thread 26892.0x7900 exited with code 3]
[Thread 26892.0x6810 exited with code 3]
[Thread 26892.0x1c3c exited with code 3]
[Inferior 1 (process 26892) exited with code 03]
(gdb) bt
No stack.
(gdb)

So, this is apparently what is causing the crash.

Ghostkeeper, 2017-05-31:
That just gave me a hint for a bug we still have: When you set the temperature to 0, the engine crashes.

cncpadawan, 2017-06-01:
WOOT!  Glad to be helpful! :)

mypeopoly, 2017-10-02:
hi Cura team, just thought you may want to know we have created a customized version of Cura for Moai that deals with this particular issue.  You can find them here: https://github.com/peopoly3d/Cura
this is was possible with help from you and cncpadawan. Thank you.

By reducing snap-distance from 1000 to 100, we were able to keep more polygons. As expected, slice time and gcode size increased.  We are happy with the results for Moai, however, we are not sure changes impact may have if it is used for regular FDM printer.

Ghostkeeper, 2017-10-12:
@mypeopoly The least we can do is pull your profile changes in. Could you make a pull request for that?

I'm not sure what snap distance you reduced. It doesn't seem to be in that Cura fork or in your [CuraEngine fork](https://github.com/peopoly3d/CuraEngine). If it's an easy one it could be made into a setting but we'd need to discuss that first because it might be too expert-ish to be exposed to the user.

mypeopoly, 2017-10-15:
I think we can do the profile pulls so they are current.  I will work with @cncpadawan to make the result. The user can still print with Standard Cura with Moai profiles; it is just that some small details may be lost.  

Since reducing snap-distance results in much longer slicing time and that we don't know how extra polygons may affect FDM printing, I recommend not making it as a setting to avoid unintentional results.

jackha, 2018-07-18:
Closed because of inactivity

closed 2018-07-18 by jackha

roseg0ld, 2018-12-03:
I am running into the CMake Error at CMakeLists.txt:23 (get_filename_component):
  get_filename_component unknown component CACHE error as well! Can anyone provide more information about how to solve it?

nallath, 2018-12-04:
You need to set the CMAKE_INSTALL_PREFIX