Ultimaker/Cura #1697: fatal exception, please help

3DPrinterPoint, 2017-04-17:
Version: 2.3.1
Platform: Windows-10-10.0.14393
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
  File "C:\Python34\lib\codecs.py", line 319, in decode
MemoryError

fieldOfView, 2017-04-17:
Try removing the log file in C:\Users\[your username]\AppData\local\cura and starting Cura again.

LipuFei, 2017-04-17:
@3DPrinterPoint Could you remove the log files in `C:\Users\[your username]\AppData\local\cura` and try again? It was trying to read a log file (too big) and got out of memory.

3DPrinterPoint, 2017-04-17:
Hello Aldo and Lipu, thank you both for the tip. It worked, Cura is running again!

closed 2017-04-17 by 3DPrinterPoint

fieldOfView, 2017-04-17:
Good to hear. AFAIK this was fixed in Cura 2.4

Ghostkeeper, 2017-04-18:
Yes, we just caught the MemoryError and continued on our merry way with upgrading the next file.