Ultimaker/Cura #1703: Cura does not create a jobname when loading a model through the commandline

fieldOfView, 2017-04-19:
When loading a model through the command line (eg ```python cura_app.py model.stl``` or by dropping a model on the Cura icon), the job name remains empty. 

Empty jobnames are a problem when saving gcode (a file named ".gcode" will become invisible on certain platforms) or sending gcode to a printer (not sure how the UM3 responds, but the OctoPrintPlugin did not like POSTing a file named ".gcode").

The command line interface is especially critical for people printing directly from Fusion 360.

Ghostkeeper, 2017-04-24:
Thanks for getting to the bottom of those issues and finding the root cause. I'm sure we can find a fix for this.

closed 2017-05-03 by nallath

Ghostkeeper, 2017-05-05:
To clarify, this issue has been fixed since then.

Wisnaeme, 2017-05-14:
I still have this problem. What was the fix for this?

fieldOfView, 2017-05-14:
It has been fixed in the source code, but a fixed version has not yet been released. Cura 2.6 beta will be the first version with the fix.

Specifically, if you are running Cura 2.5 from source and want to apply the fix yourself, this is the fix:
https://github.com/Ultimaker/Cura/pull/1781/files