Ultimaker/Cura #1704: Cura 2.5 not opening when when opening STL

kr15uk, 2017-04-19:
Cura 2.5
macOS 11.12.4

If Cura is closed and I double click on STL file, Cura icon jumps once and doesn't open.
But if Cura is running in the background, opening STL file isn't a problem.

I had Cura 2.4 initially but I've removed manually Cura.app and "cura" folder from ~/Library/Application Support to start fresh with Cura 2.5.

Thanks.

kr15uk, 2017-04-19:
Forgot to mention, nothing in cura.log file is being written by this event. Last line is when I quit Cura.

awhiemstra, 2017-04-19:
This sounds like a problem we have seen on OSX with in-place upgrades. If it is indeed the same problem, starting directly from the DMG or from a different name (e.g. Cura2.app) should work properly.

The actual solution to this seems to be to restart your computer. It seems that for some reason OSX keeps some stuff in memory that it really should not, which then conflict with the new files in the new version.

kr15uk, 2017-04-24:
Finally managed to go about and do a reboot.
Unfortunately that doesn't solves the issue.

I've also tried to define default silicer as Slic3r and then change it back to Cura so it presumably rewrites path files etc but with no success...

Happy to do some tests if required.

Thanks.

Ghostkeeper, 2017-05-02:
Since the problem does not occur on our end and it doesn't generate a log, we can really only guess at the problem.

You could try to open a terminal, and calling Cura from there:

    ~/Library/Applications/Cura2.app ~/path/to/some/model.stl

This should call Cura (if the path to the app file is correct) and immediately load the STL file. And if starting Cura normally works, then I suspect this also works. However it is not exactly the same as double-clicking on the STL file, since MacOS handles that as if you normally start Cura and immediately drop the STL onto the application, rather than supply the STL as a parameter like Linux and Windows do.

kr15uk, 2017-05-02:
An update.
Did  /Applications/Cura.app/Contents/MacOS/cura ~/path/to/some/model.stl 
Cura opened and loaded file fine.
Tried double-clicking on .stl file once again - still the same issue.

Then tried to open 3mf file via CLI, loaded fine and double-clicking worked fine as well.

Finally tried .stl again by double-click - and it works now... Bizarre, but fixed.

(3mf's wasn't opening before as well)

Thanks for your help!

closed 2017-05-03 by nallath