Ultimaker/Cura #1705: Conical Support Reversed 2.5

seantapscott, 2017-04-20:
Just upgraded to 2.5.0 from 2.4, discovered that Conical Support has been reversed.  The ToolTip description was accurate for 2.4: "Negative angles cause the base of the support to be wider than the top".  But in 2.5, layer view is rendering a non-negative 30 degree support angle as being wider at the bottom.

Ghostkeeper, 2017-04-24:
I'm not able to reproduce this. I used the default 30 degree conical support angle and it made the support narrower at the base than at the top.

seantapscott, 2017-04-24:
I would give you the file i am working on.  2.5 slices it completely differently than 2.4 .  Not sure where/how you upload here

Ghostkeeper, 2017-05-01:
You can put it in a zip-file and then drag it to the comment text area on this page.

ianpaschal, 2018-02-26:
Poster did not respond for almost a year. Closing.

closed 2018-02-26 by ianpaschal