Ultimaker/Cura #1708: Support Horizontal Expansion Setting Differs Between UM3 & UM3X

retegic, 2017-04-20:
Cura 2.5
Win 10, 64-bit

UM3X- Stable Version Firmware
AA 0.4 - UM PLA    BB 0.4 - UM Natural PVA

The issue I am experiencing happens with any combination involving PVA or UM Natural PVA(Tested with ABS, PLA, Nylon and CPE). When slicing the default settings with the UM3X profile, the support horizontal expansion is 0.2mm. When slicing default settings on the UM3 profile, the support horizontal expansion is 3mm.  I have noticed this issues since the version before 2.4. Contacted FBRC8, back when I first experienced it, I was told they relayed it to you. I wasn’t sure if they got around to it since it hasn’t been fixed. Although this is a very minor bug, it can be a tiny bit annoying.

Thanks guys!
[image]
[image]

nallath, 2017-04-24:
That's odd. It should be 3mm in both cases. I'll have a look and see what' s going on.

nallath, 2017-04-24:
I think this has been fixed for the next release, as it works when using the data from master.

Ghostkeeper, 2017-04-24:
When I updated the settings just before the 2.5 release I copied everything from UM3 to UM3E as well. But that should be working fine in the 2.5 final version, just not if he's using a 2.5 beta version.

retegic, 2017-04-24:
I was running 2.5.0 non-beta

Uninstalled all versions of Cura including 2.5, then removed all remaining files. Reinstalled 2.5 with the horizontal expansion still at 0.2mm for the UM3E 

BTW,  Norton internet security does not like cura 2.5.

Ghostkeeper, 2017-05-02:
I can't reproduce this, even when I switch to 2.5 and clear my settings. It stays properly at 3mm. Are you sure you switched the support extruder selection?

retegic, 2017-05-03:
Yes, the support extruder is set to PVA. Its a common issue we are experiencing on our computer's over here(all cura 2.5 win 10). As I said, It has been happening since before 2.4. FBRC8 was the one who confirmed it for me when the issue first arose.  Let me know if you come up with anything or need me to try some stuff.

retegic, 2017-05-23:
Fixed in 2.6 beta.

Thanks guys!

closed 2017-05-23 by retegic