Ultimaker/Cura #1714: How to make CURA run?

Henrik14, 2017-04-21:
I am brand new user of CURA, and it has one very unusuall feature, there is no RUN button?
So  ,it sometimes start and sometime dont, sometimes not even show loaded object on platform...
Iam old hand in programming and I have never seen such program. Kind of advanced AI?
Please, I am old and half blind so not much time is left for work, and I am wasting time waiting
for CURA to make up its mind to start.....

nallath, 2017-04-24:
I don't quite understand what you mean. Cura doesn't have a run button (it never had one as far as I'm aware)

Henrik14, 2017-04-24:
That EXACTLY is problem, it sometime do not want to run or start after few
hours of waiting.

Ghostkeeper, 2017-04-25:
So... the problem is that Cura doesn't start even if you click its application icon?

When you double-click Cura's icon it'll show a splash screen to indicate that it's starting up. If that doesn't happen, that sounds like a bug.

Or did you mean there is no print button that starts your print? There is only a print button if the printer is connected to your computer.

Henrik14, 2017-04-25:
Dear sir, I am estimating time to printwhile my friend would do actual
printing, so I have no printer connected.
I am using old version of CURA to be same as my friend use, si there is no
CURA icon, it us Battery file.
But while i can load model and manipulate it, there is problem of CURA
starting to calculate.
Sometimes there is no sign of slider that shows ir is working, other time
it is gray, sometimes it show  slider
as working and do nothing whole day.... I newer sav such program that has
no start to select, and if it is working
while parameters are set and restart each time parameter is changed that is
wasting computer resources,
Also as I usually do several things in paralel  as I do not like just sit
and wait for results, I think it is not working
when other programs are executed in spite of 2 core system.
I also get unusual results that is not expected, wihile printing in
different materials
For instance I am certain that ABS is half lighter than PLA and in CURA
difference is 100gr in favour of PLA
on same object.
I am inventor and working on wind turbines, so goal is to have one as light
as possible.
First one printed in PLA had 50cm diameter and weighted 5.4 Kg. so it was
unable to start. now 95cm diameter
turbine has 2024 gr. in ABS which is what I expected, but PLA say it is
1900gr. at 100% fill?
Original blades of WPS generator is 105 cm diameter and has 2kg so recent
results would be fine, if true.
Originally this small Wind Power Station is 400W at 12.5m wind. while in
comupter simulation me turbine
work even with 1m/sec Wind, but in program turbine has no weight.
On other hand it is bucket like  and would use all Wind that enters it,
compared to narrow and thin original
blades, therefore just on that account it has to be 8 times more
productive.
But my turbine is producing huge Vortex even 1000 times greater than
turbine diameter and it feed itself
from it therefore even small turbine should be able to produce 16 MW at low
Wind speed of 5m/sec.,
if it is light enough to be able to start.
I am not worried because I made protitype by hand from old plastic bucket
anmd wash bowl, and it was
4 times more effective even on 31cm diameter and at 5m/sec wind in
windtunell. If esteemed Proffesors
did not squeeze poor Shimano, I am sure results would be at least 2 times
better.
So I vas astonished that 50cm turbine made on 3D printer weight 5.4 Kg. But
it was transparent PLA and
maybe they put glass dust in.
Can You tell me whiere I could find specific weight of Filament/Plastic for
all kind of 3D printable plastic?
I need filament that is very light jet elastic, as HDPE or PP or PET.....
You have recording of CFD simulation so You can see how it would work.
Sincerely, Marijan Pollak

fieldOfView, 2017-04-25:
What version are you using exactly? Cura 15.04 is no longer supported.

In Cura 2.5, the current version, there is an option to stop automatic slicing. This adds a "Prepare" button in the lower right corner of the Cura window. The option is in the General preferences of Cura.

Henrik14, 2017-04-25:
15.04.60 if I remember Corectly. Must  use same version as my friend. Would
ULTIMAKER sponsor my work with one larger 3D printer,
at least capable of making 60cm diameter objects and similar height= Or You
do not produce printers that large?

fieldOfView, 2017-04-25:
Tell your friend to upgrade to Cura 2.5, we cannot help with Cura 15.04. Issues and questions about Cura 15.04 should go here: https://github.com/daid/LegacyCura/issues

nallath, 2017-04-25:
Don't expect any fixes for 15.04; It's 2 year old software that is no longer maintained.

Henrik14, 2017-04-25:
For some reason he dont want to.  I had newest version installed but cannot
report new .ini file to have his printer in CURA.
It is an 1000 x 1000 x 800 printer, and we are printing new Wind turbines
for test, that is at least 8 times more effective than
standard models with 3 blades, and can save the World from Global Warming,
Energy crisis, even thirst and humger.

Henrik14, 2017-04-25:
Well, I would make do with old one as I was unable to add new printer .ini
file,
Program teling me "I have no  authorization to do this" even if I run it as
Admin.
Also, I guess there are differences introduced that make old .ini files
incopatible.

Henrik14, 2017-04-30:
We would Upgrade if someone explain how it is possible to include .INI file
of new 3D printer.
Whichever way I tried, short of writing by hand about 1000 lines of thast
.ini file, was not succesfull.
I tried to copy File in folder with other already present printers and was
informed tha I lack authority
to write in that Folder. even if I have rank of Administrator. Program also
do not accept .ini files put
somewhere else. So I gave up ast last, which I seldom do.
Can You help? I can send You file if necesary.

Ghostkeeper, 2017-05-02:
There is no way to import machines from the old Cura into the new, short of writing a JSON file by hand.

You could try importing it as a profile but I don't think that would work properly with a machine file.

Ghostkeeper, 2017-05-02:
You should be able to find the specific mass of a filament on the manufacturer's page. We know the specific mass of some materials, such as the Ultimaker-brand, but for the rest Cura won't be able to give a good estimate of how heavy the print will become.

Henrik14, 2017-05-02:
Yes, i tried everypthing I can think of, but was expresly forbidden to copy
files in same directory.
What happened with backward compatbility, nobody think of that?

Henrik14, 2017-05-02:
I found list of specific weight where ABS weight is
aronud 1.0 and PLA 1.2 while PP is 0.9 which match
my estimaste from discarded plastic with that mark.
My companion has bought thanspatent PLA that
weighted in my estimation around 5, so turbine for
wind had 5.4 Kg at 50cm radius, and now same turbine
with 97  cm diameter weight 1900 gram.

Ghostkeeper, 2017-05-05:
> What happened with backward compatbility, nobody think of that?

We implemented that for profiles. The problem is that if we try to automatically translate settings from 15.04 to 2.6 there will have been so many changes to the slice process that you can never expect the result to be remotely similar in quality. For profiles that's fine, people can edit them because they are part of the user's configuration. Definitions are part of the actual Cura installation though so they need to live up to a certain quality standard. We expect that anyone who writes a printer specification is also technical enough to write a JSON document; it's not that hard.

Ghostkeeper, 2017-05-05:
Going back to the problem at hand, what would you expect a "run" button to do?

Cura can only display any button if it's already running.

Henrik14, 2017-05-05:
Sorry to say, it seems to me that You are contradicting Yourself.
If old file to You automaticaly translate to new one it can only be correct
or better in Quality than before.
Humans are incapable of handling  many changes simultaneously, that is like
learning new language.
My friend is Ph.D. for 3D printers, but He produces its own 3D printers is
bussy with everyday work.
If he vas incapable doing it, I cannot because I knew nothing about either
versions of CURA.
It is really easier to write cross compiler in style: "exchange this for
that".
It is Printer INI file we are talking about. or I misunderstood Topic of
discussion?

Henrik14, 2017-05-05:
I expect to know that slicing started, because I ordered CURA to do so.
Otherwise if it does not start Slicing for any reason, what can I do if it
wont
start?
For instance, I tried to slice object 100cm in diameter on 100 cm x 100 cm
work spac
and CURA dont want to do any thing untill I made object 97 cm in diameter.
If object cross workspace border any place (which it should not, I create
them in OPEN SCAD)
then also it would not start, and since my eyes are bad I can hardly notice
there is a problem.
So I would suggest to change colour or add function to center Object on
workspace.
But there are instances where Cura show slicing trackpad but would do
nothing more than 12 hours.
So if it do not start noticeably I have to clear platform and reload
object, but it does not work most of
time.
I usually change some parameter or two and then change them back, but it
also does not help most
of times.
I do serious work, and my hour of work is worth 15 000 $, with CURA so
unpredictable, I am loosing
lot of valuable time.

Ghostkeeper, 2017-05-10:
I understand from your context that your idea is that the "Run" button should start slicing.

We recently added this feature. Normally it automatically slices when you make any change to the print. There is now an option "Slice Automatically". It's enabled by default. When you disable it you get your "Run" button.

closed 2017-05-10 by Ghostkeeper

Henrik14, 2017-05-10:
That of course is in new version for which I do nit have .ini file for
target printer.......

Ghostkeeper, 2017-05-12:
Indeed. Implementing that feature in your version from 2 years ago is not under consideration however. Perhaps you can get that .ini file translated to the more modern .def.json files? Perhaps it suffices to use the Custom FDM Printer feature in Cura 2.5 with correct dimensions?

Henrik14, 2017-05-12:
Well, Iuse CURA only to find out hor much my turvine would weight or to
find out how much I have to pay for 3D object printed. Doss new version has
.ini file for forcebook ULTIMAKER may be? Any wax I am recinstructing one
ROSTOCK, one KOSSEL amd one VALTZER, that is 6 sided KOSSEL with sides of
123  cm that I made., There perhaps I would need to know how to write ini
files for CURA.
However when I tried to set printer of workspace 2000  x 2000 x 1000
resulting Gcode claimed to have 146 Terabytes, which I do not have on my
compuiter. I started programming on computer with 32 KBy of RAM, 252 -kBy
of virtual memory onmagnetic drum amd 2 MBy of exchangeable HDs.
 It seems to me CURA would soon require Clouds to run..
I also  noticed  that 3D printers seems to be programmed by Murphy's Law,
many short moves wherelong moves are faster, straight moves when circular
ones wiold fit much better...as with my turbines for Wind.  ...And turbines
produce huge Vortex  and feed itself by up to 1000 tomes more wimd
thasn otherwise their diameter would allow. and so 2m turbine would on Wind
of 5m/sec, produce 16 MW, that would be equivalent of standard turbine with
3 blades, 2 Km in diameter...........

Ghostkeeper, 2017-05-15:
> Doss new version has .ini file for forcebook ULTIMAKER may be?

There is no definition for Forcebook Ultra 3D yet. Not in this version or any older one.