Ultimaker/Cura #1715: exported materials do not contain cost

MarcusWolschon, 2017-04-21:
Materials exported with Cura 2.5 do not contain any information about filament cost anymore.

fieldOfView, 2017-04-21:
> do not contain any information about filament cost anymore.

They never did.
The reason for not storing the cost in the material profile is that we don't know the price a reseller asks for a material, even if we would know the MSRP of the roll of material. So we let the user fill in this information.

closed 2017-04-24 by nallath

MarcusWolschon, 2017-04-24:
So exporting materials is not intended for the user to backup his materials (e.g. before installing a new Cura version) or move them to a new computer?

fieldOfView, 2017-04-24:
It would be much less work to just copy over the whole configuration folder.

MarcusWolschon, 2017-04-24:
The user is never informed where that folder even is
and it's usually placed so it is hidden from the user
(MacOS Libraries folder or inside the .app itself or home dot-file or combinations of these depending on what Cura versions are installed, Linux dot-file, Windows app-settings/local).

fieldOfView, 2017-04-24:
Help -> Show Configuration Folder

fieldOfView, 2017-04-24:
And to answer your question:
> So exporting materials is not intended for the user to backup his materials 

You are right, it is not intended for that.