Ultimaker/Cura #1716: materials disapear until scrolling

MarcusWolschon, 2017-04-21:
In Cura 2.4 and 2.5 some custom materials are missing from the list in "manage materials" until you scroll down and scroll back up again.

Ghostkeeper, 2017-04-25:
Whoa, weird.

What operating system are you using?

Does it appear as a blank line in the material manager or is it just not listed until you scroll? If it's the first, we might be looking at a Qt bug. If it's the second, perhaps it's an update bug resulting from us asynchronously filling the list or something...?

MarcusWolschon, 2017-04-26:
Windows
They are just not listed until I scroll.
Since it's only custom materials that are affected, my guess is some update bug too. That the materials are added later but after adding them the model does not inform the view that there is an update.

fieldOfView, 2017-05-12:
See https://github.com/Ultimaker/Cura/pull/1757/commits/a727e4ffd09d365dfaa3f532c912a952f30e7089

closed 2017-11-14 by ChrisTerBeke