Ultimaker/Cura #1717: discard or keep settings dialog unusable

MarcusWolschon, 2017-04-21:
Cura 2.5
Windows 10 on a 4K display

when switching profiles the "discard or keep settings" dialog can not be used.
I assume there are some buttons off-screen that I am supposed to press.
Resizing the dialog is possible but doesn't reveal additional ui elements
nor does it completely reveal the half cut off dropdown-list.

[image]

fieldOfView, 2017-04-22:
This was fixed in master 20 days ago, but unfortunately the fix was not applied to the 2.5 branch. See https://github.com/Ultimaker/Cura/pull/1636

closed 2017-04-24 by nallath

MarcusWolschon, 2017-04-24:
Then let's set the Milestone to 2.5.1 or 2.6

fieldOfView, 2017-04-24:
I have not seen the Milestone feature being used in the Cura project.