Ultimaker/Cura #1719: Can't change filament color

SnowZA, 2017-04-21:
In the material manager of cura 2.5, cura crashes if I try to change the color of the filament. The color picker dialog opens, but as soon as I select a color, cura crashes to desktop. The log contains the following:

`--- Logging error ---
Traceback (most recent call last):
  File "D:\env\inst\Lib\logging\__init__.py", line 980, in emit
  File "D:\env\inst\Lib\logging\__init__.py", line 830, in format
  File "D:\env\inst\Lib\logging\__init__.py", line 568, in format
  File "D:\env\inst\Lib\logging\__init__.py", line 536, in usesTime
  File "D:\env\inst\Lib\logging\__init__.py", line 380, in usesTime
RecursionError: maximum recursion depth exceeded in comparison
Call stack:
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "D:/2.5/build/inst/bin/cura_app.py", line 71, in <module>
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 627, in run
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\Settings\ContainerManager.py", line 232, in setContainerMetaDataEntry
  File "C:\Program Files\Cura 2.5\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 48, in setMetaDataEntry
    container.setMetaDataEntry(key, value)
  File "C:\Program Files\Cura 2.5\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 48, in setMetaDataEntry
    container.setMetaDataEntry(key, value)
  File "C:\Program Files\Cura 2.5\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 48, in setMetaDataEntry
    container.setMetaDataEntry(key, value)
<<snip>>
  File "C:\Program Files\Cura 2.5\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 43, in setMetaDataEntry
    super().setMetaDataEntry(key, value)
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Settings\InstanceContainer.py", line 210, in setMetaDataEntry
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Logger.py", line 58, in log
  File "C:\Program Files\Cura 2.5\plugins\FileLogger\FileLogger.py", line 36, in log
    self._logger.warning(message)
  File "D:\env\inst\Lib\logging\__init__.py", line 1291, in warning
  File "D:\env\inst\Lib\logging\__init__.py", line 1415, in _log
  File "D:\env\inst\Lib\logging\__init__.py", line 1425, in handle
  File "D:\env\inst\Lib\logging\__init__.py", line 1487, in callHandlers
  File "D:\env\inst\Lib\logging\__init__.py", line 855, in handle
  File "D:\env\inst\Lib\logging\__init__.py", line 1048, in emit
  File "D:\env\inst\Lib\logging\__init__.py", line 986, in emit
Message: 'UM.Settings.InstanceContainer.setMetaDataEntry [210]: Meta data with key color_code was not found. Unable to change.'
Arguments: ()
2017-04-21 17:25:18,228 - WARNING - UM.Settings.InstanceContainer.setMetaDataEntry [210]: Meta data with key color_code was not found. Unable to change.`

fieldOfView, 2017-04-21:
Which material did you try? Was it duplicated from one of the materials that show up with a white "swatch" in the list but a yellow swatch in the selected material view on the right?

Devs: The problem is that these materials define no color_code in the XML file, as reported here:
https://github.com/Ultimaker/fdm_materials/pull/5#issuecomment-293696004

SnowZA, 2017-04-24:
Yes, it was from one of those. I was trying to make my own 1.75mm materials, so duplicated one that already had a 1.75mm diameter.

fieldOfView, 2017-04-24:
As a workaround, duplicate one that already has a working color instead.

Ghostkeeper, 2017-04-25:
I just added the missing colour codes: https://github.com/Ultimaker/fdm_materials/commit/728ba51536b66fdd289898e6d153af089da30e34

Ghostkeeper, 2017-04-25:
I'm not able to reproduce this, by the way, not even before I added those colour codes.

fieldOfView, 2017-04-25:
I could. I think the actual crash may very well be platform specific, since Qt tries to open a system color picker.

SnowZA, 2017-04-25:
I'm using this on Windows 10. The crash happens when I try to close the system color picker. The dialog opens fine, but the moment it tries to update the material with the selected color, the crash happens.

BagelOrb, 2017-07-23:
This bug should be fixed now.

closed 2017-07-23 by BagelOrb