Ultimaker/Cura #1720: Releases Time Alert?

filipgoc, 2017-04-21:
Is there any place where we can see if a next release (2.6 2.7...) is coming up soon, which makes Cura closed to adding new stuff? 

I know there's nothing coming super soon, I'm curious in general.

fieldOfView, 2017-04-21:
AFAIK, the aim is to have a release every 2 months (since Cura 2.4), so Cura 2.6 should be released end of june. The beta period has taken around a month for Cura 2.4 and 2.5, so the open beta for Cura 2.6 is likely due in a month. I would guestimate a feature-freeze for Cura 2.6 is about half a month away.

filipgoc, 2017-04-21:
That's informative. Thanks.

fieldOfView, 2017-04-21:
Note that it is based on observation; I am no more privy to these things than you are.

filipgoc, 2017-04-21:
Ok. Noted.

Appesteijn, 2017-04-22:
We are currently trying to deliver a releasable version every sprint (two weeks) and to release every 4 sprints. We will create an internal 2.6-BETA on 4 May, public 2.6-BETA is planned for 18 May. If all goes well and we don't find a lot of bugs 2.6-final is planned to be released 1 June.

Please bare in mind we are changing a lot in our way of working and unforeseen obstacles keep popping up so these dates might change.

filipgoc, 2017-04-22:
Awesome info @Appesteijn . We understand it's just a hint; of course things change all the time at these paces.

closed 2017-07-23 by BagelOrb