Ultimaker/Cura #1721: feature request: auto reload on file change

Ognian, 2017-04-21:
Hello,
an option (check box) in the view to autoreload when checked  would be great e.g. when  the underlying .stl file changes then the view is refreshed (at the moment it has to be triggered manually with F5). This will improve a lot the overall workflow...
Thanks a lot
Ognian

nallath, 2017-04-24:
Thanks for the feature request! I've created a ticket for it (CURA-3716). We will let you know if it's going to be implemented (which is up to our product owner).

Ghostkeeper, 2017-04-25:
Does Python have functionality to monitor the file system or do we need an external library for that?

awhiemstra, 2017-04-25:
I don't know if Python has it, but I do know that Qt has it. So we won't need an extra dependency in any case.

jackha, 2018-07-18:
Deferred because it ended up too low of our backlog.

Ghostkeeper, 2018-08-30:
This has been implemented now.

closed 2018-08-30 by Ghostkeeper

bo0tzz, 2020-03-21:
Hi! I've been looking for this feature but can't find it anywhere. Is it still present?

fieldOfView, 2020-03-21:
It is, but it is disabled for Windows because the feature caused a crash on Windows.

bo0tzz, 2020-03-21:
I'm running on linux, but I can't find it. Any pointers?

Ghostkeeper, 2020-03-24:
Once a file on the build plate changed, you should see a message pop up on the bottom side of the screen, asking you whether you want to reload your mesh.

It currently doesn't work for meshes that contain multiple submeshes (like AMF and 3MF) but that's something that we're improving in Cura 4.6.

wohltat, 2021-02-24:
There is a manual reload but no "auto reload" in linux. Or am i missing something?
Would love to see that feature in cura.

Ghostkeeper, 2021-03-03:
It's not 100% automatic. That would be inadvisable because people would normally expect the data loaded in Cura to not change once it's loaded.
It's implemented with this pop-up:
[image]
So you still have to press the button to reload it.