Ultimaker/Cura #1722: Cura 2.5 missing graphics for icons

mdiorio, 2017-04-22:
Windows 10 on Microsoft Surface Pro 3 and the icon images are missing everywhere.

[image]

fieldOfView, 2017-04-22:
We have not seen that in a while...
Could you check if the SVG files are there in the first place, just to make sure this is not an installer issue?
C:\Program Files\Cura 2.5\resources\themes\cura\icons

mdiorio, 2017-04-22:
All the SVG files are there and valid.

fieldOfView, 2017-04-22:
Thanks. How much RAM do you have on your SP3?

mdiorio, 2017-04-22:
4GB.  I can run things like AutoCad and Fusion 360 on it without an issue.

fieldOfView, 2017-04-22:
I was not suggesting anything like "shame on you for trying to run Cura with only 4Gb" (  ;-)  ), but more trying to figure out how we can try to reproduce this. Obviously it has not cropped up during testing on all Windows 10 machines, or it would probably not have been released.

mdiorio, 2017-04-22:
Didn't think that at all. More just saying that it probably isn't memory.

It happened on the last build too. I didn't get around to reporting it
though.

fieldOfView, 2017-04-22:
> More just saying that it probably isn't memory.

With Cura, you never know ;-)
Hopefully we can reproduce it, and fix it.

fieldOfView, 2017-04-22:
By "previous build", do you mean Cura 2.4 or Cura 2.5 beta? Did you ever use previous versions of Cura 2?

mdiorio, 2017-04-22:
2.5 Beta. I can install 2.4 to test. I'm pretty sure I did and it worked.

mdiorio, 2017-04-22:
Works perfectly in 2.4

Ghostkeeper, 2017-04-25:
I'm obviously not reproducing this on Ubuntu and neither is our QA team or half of our developers on Windows 10. I'm going to create an issue for this so that we can plan/track the problem, but I expect that it won't be picked up any time soon because we don't have the Surface here at work.

BagelOrb, 2017-05-12:
Another user has reported the same bug: https://www.reddit.com/r/3Dprinting/comments/6aj09m/cura_25_doesnt_have_icons_just_various_grey_boxes/

mdiorio, 2017-05-12:
Glad it's not just me. I installed it on a Windows 10 non-surface device
and it works fine.

amoose136, 2017-05-15:
I have the same issue currently. I have 14GB of RAM.

sisu-jason, 2017-05-26:
We are having a similar issue with both Cura 2.5 and 2.6 BETA (just installed both yesterday for the first time). 2.4 worked fine, installed 2.5 and the icons don't show. SVG files are there and I have tested one (opened with Inkscape). I currently have 2.4, 2.5, and 2.6 installed; icons only work in 2.4.

This is on a Lenovo desktop, Windows 10 Pro 64 bit, core i7 CPU, 16 GB ram, Nvidia K2000D graphics card

awhiemstra, 2017-05-29:
I suspect this is due to changes in the shader responsible for rendering the icons. Having some log output would be useful here. However, it might be the Cura log does not show anything and you need the Windows debug log which is trickier to get at.

mdiorio, 2017-05-29:
If you just need to attach windbg to the spawned Cura process I can do
that.

sisu-jason, 2017-06-07:
Just a little update that may help: Today we installed Cura 2.5 on a computer of similar specs to the first one (Lenovo desktop, windows 10), but it had never had any version of Cura installed previously and 2.5 worked fine; all icons there. The differences between the computers that may be of interest:
The first computer (2.5 without working icons) 
1. had Cura 2.4 on it previously
2. has python 2.7 installed as well.

Not sure if either of those things would/could affect the 2.5 installation, but maybe?

nallath, 2017-06-08:
I don't think that any of those will have an influence on this.

sisu-jason, 2017-06-20:
Here is a log file for Cura 2.6 if it helps. The icons still don't show up; neither does the webcam feed. Other than that I can use 2.6 just fine.

`2017-06-20 09:47:53,833 - DEBUG - UM.Qt.QtApplication.windowClosed [198]: Shutting down cura
2017-06-20 09:47:57,834 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
2017-06-20 09:47:57,834 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [203]: Killing engine process
2017-06-20 09:47:57,838 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [206]: Engine process is killed. Received return code 1
2017-06-20 09:48:37,834 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin FileLogger
2017-06-20 09:48:37,840 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin GCodeProfileReader
2017-06-20 09:48:37,849 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin GCodeReader
2017-06-20 09:48:37,856 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin GCodeWriter
2017-06-20 09:48:37,866 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin ImageReader
2017-06-20 09:48:37,882 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin LayerView
2017-06-20 09:48:37,888 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin LegacyProfileReader
2017-06-20 09:48:37,898 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin LocalFileOutputDevice
2017-06-20 09:48:37,907 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin MachineSettingsAction
2017-06-20 09:48:37,961 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin OctoPrintPlugin
2017-06-20 09:48:37,984 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin PerObjectSettingsTool
2017-06-20 09:48:37,992 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin PostProcessingPlugin
2017-06-20 09:48:38,004 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin RemovableDriveOutputDevice
2017-06-20 09:48:38,040 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin SliceInfoPlugin
2017-06-20 09:48:38,046 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin SolidView
2017-06-20 09:48:38,056 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin CameraTool
2017-06-20 09:48:38,067 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin MirrorTool
2017-06-20 09:48:38,080 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin RotateTool
2017-06-20 09:48:38,096 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin ScaleTool
2017-06-20 09:48:38,106 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin SelectionTool
2017-06-20 09:48:38,115 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin TranslateTool
2017-06-20 09:48:38,130 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin UltimakerMachineActions
2017-06-20 09:48:38,163 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin UM3NetworkPrinting
2017-06-20 09:48:38,199 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin._onServiceChanged [203]: Bonjour service added: ultimakersystem-ccbdd3002925._ultimaker._tcp.local.
2017-06-20 09:48:38,222 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin UpdateChecker
2017-06-20 09:48:38,223 - INFO - UpdateChecker.UpdateCheckerJob.run [43]: Checking for new version of cura
2017-06-20 09:48:38,249 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin USBPrinting
2017-06-20 09:48:38,263 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin VersionUpgrade21to22
2017-06-20 09:48:38,271 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin VersionUpgrade22to24
2017-06-20 09:48:38,282 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin VersionUpgrade25to26
2017-06-20 09:48:38,291 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin SimpleView
2017-06-20 09:48:38,306 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin X3DReader
2017-06-20 09:48:38,318 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin XmlMaterialProfile
2017-06-20 09:48:38,326 - INFO - UM.PluginRegistry.loadPlugin [176]: Loaded plugin XRayView
2017-06-20 09:48:38,327 - INFO - UM.Qt.QtApplication.__init__ [112]: Command line arguments: {'external-backend': False, 'disable-textures': False, 'single_instance': False, 'file': [], 'qmljsdebugger': None}
2017-06-20 09:48:38,338 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-06-20 09:48:38,338 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [311]: Attempting to connect to COM3
2017-06-20 09:48:38,339 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [311]: Attempting to connect to COM1
2017-06-20 09:48:38,347 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\Ultimaker+3+Extended_current_settings.inst.cfg to version 2000001.
2017-06-20 09:48:38,354 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_extended_extruder_left+%232_current_settings.inst.cfg to version 2000001.
2017-06-20 09:48:38,361 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_extended_extruder_right+%232_current_settings.inst.cfg to version 2000001.
2017-06-20 09:48:38,371 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker+3+extended_draft+print+-+sisu+custom+-+0.3mm.inst.cfg to version 2000001.
2017-06-20 09:48:38,378 - INFO - UM.VersionUpgradeManager._upgradeFile [353]: Upgraded .\ultimaker3_extruder_left_draft+print+-+sisu+custom+-+0.3mm.inst.cfg to version 2000001.
[... cut ...]
2017-06-20 09:50:00,591 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-06-20 09:50:00,592 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-06-20 09:50:14,932 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-06-20 09:51:05,106 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles`

TURTLEHARDBODY, 2017-07-17:
I have the same issue. 
    i5 4670 @ 3.4 ghz
    Asus - H87-PLUS ATX LGA1150 Motherboard
    4x4 GB ram (16 GB)
    Samsung 840 Evo SSD
    MSI Geforce GTX 1070

smat6177, 2017-07-22:
I got the same problem too.

Ghostkeeper, 2017-08-08:
Our project manager just rejected this bug because it's more than 3 months old.

Ghostkeeper, 2017-08-08:
Is this still a problem using the 2.7 beta? We switched to a newer version of Qt.

sisu-jason, 2017-08-08:
Yes, this is still an issue. I downloaded 2.7 BETA and the same issue still
exists; no icons (or video feed), but everything else works...

-Jason

Jason Dearden
Mechanical Engineer
Sisu Devices
1520 Royston Lane
Round Rock, Texas 78664
512.377.6075 ex 235
<http://www.sisudevices.com>www.sisudevices.com

debsahu, 2017-08-08:
Please dont close this issue, still the icons show up as black boxes. Running this on a windows 10 PC with 16+GB of RAM.

mdiorio, 2017-08-08:
Yup. Same here. Still and issue on my surface. Works fine on my Dell
Windows 10 laptop though.

amoose136, 2017-08-15:
After fully reformatting it works on my system now. Interestingly the thing that caused me to reformat was that windows update got broken after some registry edits to fix another problem so it was a several month old build of windows 10 I was using. I wonder what build numbers people are at.

Ghostkeeper, 2017-08-23:
> Please dont close this issue, still the icons show up as black boxes. Running this on a windows 10 PC with 16+GB of RAM.

While our project manager removes it from the internal tracker and therefore from our planning, I like to leave the issues open on Github to make it public that these issues still exist and hopefully to prevent getting duplicate reports.

Todderbod, 2017-09-03:
Im having the same problem just update to 2.7 beta and still no icons..any fix for it without totally re-doing my computer

21347, 2017-10-03:
Well, I just started Cura on my Windows 7 Box and saw that error (Version 2.7.0).

I didn't get any errors in the logfile pointing to any graphics problems, so I suspected some SVG-loading/scalling/whatever related issue. It turns out that replacing all SVG-Icons by PNG files will make them show up correctly.

[image]

Looks like one cannot upload .zip files here, so here's how to create the Theme:

- Navigate to your Cura directory, i.e. "C:\Program Files\Cura 2.7\resources\themes" and copy the "cura" folder to %userprofile%\AppData\Roaming\cura\2.7\themes\cura-png
- Edit "theme.json" to give your theme a descriptive name, e.g. "Ultimaker (PNG)"
- convert all SVG files to PNG files, e.g. using inkscape and something like:
`for %%p in (*.svg) do "C:\Program Files\Inkscape\inkscape.exe" -z %%p -e %%~np.png`

As this might provide a workaround, it doesn't explain why it fails to load SVG files int the first place...

debsahu, 2017-10-03:
@21347 Wow genius. I had to delete svg files after png files were created as the last step.

[image]

21347, 2017-10-04:
> I had to delete svg files
a, crap - yes, of course - I forgot to mention this. Updated my comment, thanks @debsahu .

pscrespo, 2017-10-28:
@21347 Thank you so much! I've been having this problem since 2.5 (3.0.3 too) and this solved it! I'm using windows 10 64bit and had no issues before 2.5...

Ghostkeeper, 2017-11-06:
> EDIT: Actually, I think I found a less-hacky solution: in Windows configuration, I changed the SVG files to open with Edge instead of Internet Explorer by default. Then restored the svg files un Cura folder and it worked. Can someone else confirm if this works?

If that works it'd be very interesting since our SVGs are rendered by Qt, not by a browser. Maybe Qt uses some system library instead?

mdiorio, 2017-11-06:
Doesn't work for me. Changed default svg app to Edge and opened Cura, still
not showing the images.

Max-Headroom, 2017-12-02:
@21347. Thanks for that. Does the job.
And I used this to convert.
http://svgtopng.com/

@developers.
It seems to me what I can pick up is a Windows problem for people who upgraded to Windows 10. Something went wrong. Most people complained not being to able view SVG in I.E., Edge and Chrome. OK, I know Cura is not web based.
Maybe you should just move away from SVG to PNG?

fieldOfView, 2017-12-02:
> Maybe you should just move away from SVG to PNG?

No. SVG is scalable, PNG is not. Using PNG instead of SVG leads to less flexibility and a decrease in visual fidelity for most users. Note that there are many, MANY users of Windows 10 that are not affected by this issue (me included, but not exclusively), and SVG works just fine for them. 

Instead, the cause for this issue should be figured out. What is causing this issue on affected systems? What is the common factor between these Windows 10 installs? So far none of the Cura dev team has found a machine that is affected, making it very hard to figure out what is going on.

Max-Headroom, 2017-12-02:
@fieldOfView Got you. Pain of being a developer.
I just browsed around today and did not make notes. Another thing somebody mentioned is that he has lost all his fonts when upgrading to W10, and then his SVG also stopped working (in browsers). Maybe this can help.

It might be helpful if other Cura users can comment if they upgraded from an older Windows to Windows 10.

oli71, 2017-12-12:
I have the same issue. Using Windows 7, using Cura 2.5 and now 3.1. I have the files in the light directory. I not checked to go over to png's.
I have a lot of other software installed as well, none having such issues.
By the way i have cPython 2.7.12 installed as well. When looking at the messages here it could be a hint for a sollution i think.
By the way, this is still an issue with 3.1 i changed over today from 2.5.
Hope this helps somewhat. At least it's not the OS. - Can't do much testing on the computer i have here in use becasue it's a company managed one.

stuartpb, 2017-12-19:
I have this problem too: if memory serves, I upgraded to Windows 10 from Windows 8.

stuartpb, 2017-12-19:
Could it be a missing / conflicting dependency issue? This issue is six years old, but it indicates that that's a sort of problem that can disrupt QT SVG rendering: https://forum.qt.io/topic/3563/no-icons-when-deploy-application-using-svg-files

fieldOfView, 2017-12-19:
Nice find. The post you found is about Qt4 though, and Cura uses Qt5 which is substantially different. Just to follow up on your hunch though, here's something to try.
[Qt5Xml.zip](https://github.com/Ultimaker/Cura/files/1573180/Qt5Xml.zip)
That zip contains a dll. Could you place the dll in the main Cura folder (which already contains Qt5Svg.dll) and try if that helps with the SVG icons?

stuartpb, 2017-12-19:
Nope, dropped it into `C:\Program Files\Ultimaker Cura 3.1` and re-launched, still nothing.

stuartpb, 2017-12-19:
Just noticed that my `QT_PLUGIN_PATH` environment variable is set to `C:\Program Files (x86)\OpenShot Video Editor\`, let me try deleting that...

stuartpb, 2017-12-19:
Yep, un-setting `QT_PLUGIN_PATH` in Environment Variables from System Properties (Win+Pause -> "Advanced system settings" -> "Environment Variables...") fixed this issue for me. Is that something that can be handled upstream, or is Cura pretty much just at the whims of the run environment on this?

fieldOfView, 2017-12-19:
Awesome, some success at last. From http://doc.qt.io/qt-5/deployment-plugins.html:
> Note: QT_PLUGIN_PATH should not be exported as a system-wide environment variable since it can interfere with other Qt installations.
So it is Openshot Video Editor (in your case) that is at fault here. Hopefully a workaround can be found to ignore or dynamically unset the environment variable.

stuartpb, 2017-12-19:
I think the real solution should be to make this one of the first things to check when an issue like this comes up, and tell end users to unset the environment variable if it's been unintentionally set like this (and to further complain to the vendor that needlessly set it).

Max-Headroom, 2017-12-20:
@stuartpb
Thanks for this. had the same problem with Openshot video editor.
Sorted

oli71, 2017-12-20:
stuartpb is right on the environment variable QT_PLUGIN_PATH . I had the same setting to OpenShot v2.07. When removing the variable it works fine. With it I uninstalled Openshot and installed Openshot v2.4.1 which is the current up to date version. Both programs work well, and the QT_PLUGIN_PATH is not set with the new install.
It seems to me, the OpenShot team already fixed this.
None the less, I think this was hard to find and I am very glad having this fixed now.
Thank you.

stuartpb, 2017-12-21:
Indeed, newer OpenShot installers actually [*unset*](https://github.com/OpenShot/openshot-qt/blob/master/installer/windows-installer.iss#L86) the `QT_PLUGIN_PATH` variable, which is also a kind of intrusive action, but at least it's one that doesn't break other programs at rest.

Ghostkeeper, 2018-03-28:
As far as I know we haven't upgraded the Qt version since 2.7 so I'd expect that this bug still occurs. Maybe stuartpb or Max-Headroom or someone else that got this bug could test this again with Cura 3.2 (or 3.3) to verify?

ChrisTerBeke, 2018-03-28:
I interpreted the comments above as being an issue with another software that uses Qt and sets that environment variable incorrectly. I don't see how Cura could fix that?

debsahu, 2018-03-28:
Openshot (not Cura) was causing this issue, after removal and installing upgrade the QT_PLUGIN_PATH is fixed. Once fixed, Cura icons show up correctly.

makeitdad, 2018-04-02:
Just wanted to say that I had the same issue, and this thread on this issue resolved it for me.  Simply deleting the QT_PLUGIN_PATH system wide variable resolved it.  Thank you all!

closed 2018-04-26 by Ghostkeeper