Ultimaker/Cura #1724: Cura 2.5 can't load G-code file

sjc3f, 2017-04-22:
Did someone succeed? I always get the wrong message.(Failed to load d:\n_shell360.gcode)

nallath, 2017-04-24:
Could you send us the logs?

sjc3f, 2017-04-24:
DEBUG - GCodeReader.GCodeReader.read [220]: Preparing to load C:/Users/jcsu/Desktop/N_SHELL360.gcode
2017-04-24 23:32:06,871 - DEBUG - GCodeReader.GCodeReader.read [231]: Opening file C:/Users/jcsu/Desktop/N_SHELL360.gcode
2017-04-24 23:32:06,917 - DEBUG - GCodeReader.GCodeReader.read [254]: Parsing C:/Users/jcsu/Desktop/N_SHELL360.gcode...
2017-04-24 23:32:06,918 - ERROR - UM.Logger.logException [74]: Exception: Exception occurred while loading file C:/Users/jcsu/Desktop/N_SHELL360.gcode
2017-04-24 23:32:06,920 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-04-24 23:32:06,921 - ERROR - UM.Logger.logException [78]:   File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\FileHandler\ReadFileJob.py", line 61, in run
2017-04-24 23:32:06,922 - ERROR - UM.Logger.logException [78]:   File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshFileHandler.py", line 28, in readerRead
2017-04-24 23:32:06,924 - ERROR - UM.Logger.logException [78]:   File "C:\Program Files\Cura 2.5\plugins\GCodeReader\GCodeReader.py", line 301, in read
2017-04-24 23:32:06,925 - ERROR - UM.Logger.logException [78]:     current_position = self._processGCode(G, line, current_position, current_path)
2017-04-24 23:32:06,926 - ERROR - UM.Logger.logException [78]:   File "C:\Program Files\Cura 2.5\plugins\GCodeReader\GCodeReader.py", line 194, in _processGCode
2017-04-24 23:32:06,928 - ERROR - UM.Logger.logException [78]:     e = float(item[1:])
2017-04-24 23:32:06,929 - ERROR - UM.Logger.logException [78]: ValueError: could not convert string to float: 'XTRUDED'

fieldOfView, 2017-04-24:
I think we're going to need the gcode as well. Can you post it somewhere?
I don't think the gcode file was produced with Cura, was it?

sjc3f, 2017-04-24:
It produced with Cura 2.5
There are stl, g-code, log in the link.
[https://drive.google.com/open?id=0B6fNLelEmB1jcUxnMHZuZkg3Yms](url)
I use Delta printer(Elliptic build plate), Is this caused by this?

fieldOfView, 2017-04-25:
Here's a hunch: In your start gcode you have two G92 lines. Both of them have a comment with the word "extruded" in them. I think the issue will go away if you remove the "e" from "extruded" (or make it say "_extruded"). Cura trips up because of the "e" at the start of a word in the comment.

I'll get that fixed for Cura 2.6 (unless someone beats me to it)

fieldOfView, 2017-04-25:
Hmm... The provided gcode file loads just fine for me in the released version of Cura 2.5.0

nallath, 2017-04-25:
I fixed the issue right before we released the 2.5.0 final :)

fieldOfView, 2017-04-25:
Awesome!
@sjc3f, can you confirm you are still running the beta? If so please update to the 2.5.0 final release

sjc3f, 2017-04-25:
I download Cura 2.5 at 4/20, It's should be final release, but I just re-downloaded, reinstall Cura 2.5 and the problem is solved, Thanks.

closed 2017-04-25 by sjc3f

smartavionics, 2017-04-25:
Using 2.5 from the repo, I still have this problem. What's the sha1 of the fix that was applied?

smartavionics, 2017-04-25:
Found it, it's 558b446b4a44ec82102788a68aeec2b1315f4064 but that is on the master branch and not in the 2.5 branch!

nallath, 2017-04-26:
Whoops. My bad. I must have remembered wrong.

smartavionics, 2017-04-26:
No problem, at least the fix made it into the 2.5 release even if it's not in the 2.5 branch.

Ghostkeeper, 2017-05-02:
Well I do hope we built the 2.5 release with the actual 2.5 branch, not with master.