Ultimaker/Cura #1727: Can't Zoom in at object level

Nightshadep, 2017-04-22:
For some reason, the new version will not let me zoom to the object.  I can zoom in directly down on the object but not in at front or side views.

nallath, 2017-04-24:
Center camera when object is selected is disabled because we had an issue on windows. I expect it to be resolved for the next release.

What do you mean by not on front /side views? You can't zoom if you translated the camera?

Nightshadep, 2017-04-24:
What I many was in past versions I could s
son in at bed level.  And I can no lo get do that.  Any idea when next
version will be out

Also for some reason the other version keeps forgetting my pro yet and
resetting my settings.  Any idea why it would do this it is really a
problem.

Ghostkeeper, 2017-04-25:
The next version is planned for release somewhere in June. We aim to release every 2 months and we've just had the 2.5 release. A beta version will be available in a week or two though.

My workaround for this is to rotate the camera 90 degrees, move the *origin* of the camera a bit further back by dragging the middle mouse button (shifting the camera sideways), then rotating 90 degrees back. The problem is that there is a maximum zoom level and an origin that's somewhere in front of the model/buildplate/detail you want to see. So you can move the origin backwards a bit and then you can zoom in closer.

BagelOrb, 2017-07-23:
Zooming in at specific features of an object could be done by placing the origin automatically under the mouse pointer at the distance where you are pointing at the object.
The scroll in mouse pointer direction could be improved such that it places the origin at the object you are pointing to.

nallath, 2017-07-24:
The center camera issue has been resolved for the next release.

closed 2017-07-24 by nallath