Ultimaker/Cura #1728: Cura 2.5 skips lines when spiralizing

ioannisnousias, 2017-04-23:
I'm seeing a weird behavior when spiralizing a  simple 20x20x20 test cube with text on the sides. Cura 2.5 does a weird jump and ends up skipping a section, or at least that's what the 'layers' visuals shows:
[image]

It also draws a line that's totally wrong:
[image]

if instead I disable spiralize outer contour and use 1 line wall and 0% infill, the errors are gone, but the travel paths are pretty crazy:
[image]

snapshots show 0.2mm layer height, but the issue appears with the default 'normal quality' settings too.

smartavionics, 2017-04-23:
Hi, I would like to investigate this, can you please make the model available.

smartavionics, 2017-04-23:
Just had a thought, is the model solid? Spiralization works best when the model is solid.

smartavionics, 2017-04-23:
I have created a solid model similar to yours and it still has a problem so I will look into that.
[image]

ioannisnousias, 2017-04-23:
Sorry, I just saw your comments. Thanks for your prompt response. I see you've already replicated the problem, but here is my model for completeness:
[cube20x20x20-labels.zip](https://github.com/Ultimaker/Cura/files/949593/cube20x20x20-labels.zip)

The object is solid, I just used 0% infill above.

smartavionics, 2017-04-23:
Thanks for the model.

My thoughts so far... 

There will be a problem printing this even if it spiralized perfectly due to the lack of support at the bottom of the X you can't spiralize out into thin air! Models that are good for spiralization should not have shallow gradients (surfaces should slope at least 45 degrees).

The diagonal line that joins the top corner of the X to the corner of the cube is not easy to eradicate because what's happening there is that on one layer there is more detail (the X cutout) than the layer above. The spiralization algorithm tries to create a smooth spiral by starting the next layer at a vertex that is close to the end vertex of the previous layer but without jumping backwards. So in this case one layer is finishing at the top of the X and the closest vertex on the layer above is all the way over in the corner so you get the diagonal line. I could imagine a fix for this whereby you add an extra vertex into the layer above that is closer to the last vertex in the preceding layer but the difficulty is determining when to add that vertex and where to put it! I shall think about this.
Actually, if the X did not come to an abrupt halt at the top/bottom but was blended back into the surface it would probably spiralize just fine. 
There also appears to be an issue with gaps in a couple of places which I don't understand yet so that needs further investigation.

smartavionics, 2017-04-23:
Well, I believe I have fixed the weird gaps and, by sheer chance, that has also removed the diagonal line as we see here but I think that's just by good luck and can't be guaranteed. I shall test this some more and submit a fix for 2.6. The bug I fixed should only affect models which have uniform cross section like the example here. Models that are more curvaceous (great word!) are probably not going to be affected.

Thanks again for the report.
[image]

smartavionics, 2017-04-23:
Fix for the 2.5 branch submitted.

https://github.com/Ultimaker/CuraEngine/pull/503

ioannisnousias, 2017-04-23:
Great and prompt response! You've already gota fix, but I'd like to add that this does not happen with Cura 2.4, so it's a regression. That may give you more clues as to what has happened. Thanks again :)

smartavionics, 2017-04-23:
Yes, I contributed a new implementation of the spiralizing code for 2.5 and apart from the odd glitch like this, it's working a lot better than the old code. Today's bug fix should make it into the next release (2.6).

ioannisnousias, 2017-04-23:
Sorry, out of topic question: Is there something like a feature request portal for Cura? You know how there is 'infill layer thickness'? Could you guys also add 'support layer thickness/height' ? I find that support can be too hard to remove and rigid at times (depends on the material), so I tend to set 'support line width' to 0.2 or 0.1 on a 0.4 nozzle, and I think doubling the support layer height will also make it more 'fidgety', thus easier to crumple and remove.

smartavionics, 2017-04-23:
For feature requests, just create a new issue here and the cura developers will notice and comment on what you have suggested. It looks like they are currently working on some new stuff related to supports so now is a good time to mention it.

Ghostkeeper, 2017-04-25:
Support layer thickness in particular has been requested quite often so we'd close it as a duplicate :wink: It's also in our top 50 of issues.

I think Smartavionics' fix is acceptable, given that this has always been a problem. We can devise a better solution with introducing new vertices later, but it's not trivial anyway.

Great analysis, Smartavionics. It's great to have a contributor like you in our project.

BagelOrb, 2017-06-27:
Spiralize is fixed now

closed 2017-06-27 by BagelOrb