Ultimaker/Cura #1729: 2.5 crash after clear table and open new model

Acket80, 2017-04-23:
Copy from "Ой!" window
Version: 2.5.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.5/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 653, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 216, in event
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Event.py", line 187, in call
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 305, in __performEmit
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 124, in calculateBoundingBoxMesh
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 581, in getBoundingBox
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Scene\SceneNode.py", line 648, in _calculateAABB
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 176, in getExtents
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 251, in getConvexHullVertices
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 243, in getConvexHull
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 235, in _computeConvexHull
  File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Mesh\MeshData.py", line 365, in approximateConvexHull
  File "scipy\spatial\qhull.pyx", line 2385, in scipy.spatial.qhull.ConvexHull.__init__ (scipy\spatial\qhull.c:24141)
  File "scipy\spatial\qhull.pyx", line 434, in scipy.spatial.qhull._Qhull.__init__ (scipy\spatial\qhull.c:5799)
scipy.spatial.qhull.QhullError: qhull precision warning: 
The initial hull is narrow (cosine of min. angle is 1.0000000000000002).
Is the input lower dimensional (e.g., on a plane in 3-d)?  Qhull may
produce a wide facet.  Options 'QbB' (scale to unit box) or 'Qbb' (scale
last coordinate) may remove this warning.  Use 'Pp' to skip this warning.
See 'Limitations' in qh-impre.htm.
QH6114 qhull precision error: initial simplex is not convex. Distance=-1.4e-14

While executing:  | qhull i Qt
Options selected for Qhull 2015.2.r 2016/01/18:
  run-id 857820053  incidence  Qtriangulate  _pre-merge  _zero-centrum
  _max-width 1.5e+02  Error-roundoff 2.1e-13  _one-merge 1.5e-12
  _near-inside 7.3e-12  Visible-distance 4.2e-13  U-coplanar-distance 4.2e-13
  Width-outside 8.4e-13  _wide-facet 2.5e-12  _narrow-hull -2.2e-16

precision problems (corrected unless 'Q0' or an error)
      1 flipped facets

The input to qhull appears to be less than 3 dimensional, or a
computation has overflowed.

Qhull could not construct a clearly convex simplex from points:
- p5(v3):     0     0    -0
- p792(v2):     0 -1.4e-14    -0
- p116(v1): 1.1e+02     0 -1.5e+02
- p0(v0):     0     2 -1.5e+02

The center point is coplanar with a facet, or a vertex is coplanar
with a neighboring facet.  The maximum round off error for
computing distances is 2.1e-13.  The center point, facets and distances
to the center point are as follows:

center point    26.75      0.5      -76

facet p792 p116 p0 distance= -3.6e-15
facet p5 p116 p0 distance= -3.4e-15
facet p5 p792 p0 distance=  -27
facet p5 p792 p116 distance=  -22

These points either have a maximum or minimum x-coordinate, or
they maximize the determinant for k coordinates.  Trial points
are first selected from points that maximize a coordinate.

The min and max coordinates for each dimension are:
  0:         0       107  difference=  107
  1:  -1.388e-14         2  difference=    2
  2:      -152        -0  difference=  152

If the input should be full dimensional, you have several options that
may determine an initial simplex:
  - use 'QJ'  to joggle the input and make it full dimensional
  - use 'QbB' to scale the points to the unit cube
  - use 'QR0' to randomly rotate the input for different maximum points
  - use 'Qs'  to search all points for the initial simplex
  - use 'En'  to specify a maximum roundoff error less than 2.1e-13.
  - trace execution with 'T3' to see the determinant for each point.

If the input is lower dimensional:
  - use 'QJ' to joggle the input and make it full dimensional
  - use 'Qbk:0Bk:0' to delete coordinate k from the input.  You should
    pick the coordinate with the least range.  The hull will have the
    correct topology.
  - determine the flat containing the points, rotate the points
    into a coordinate plane, and delete the other coordinates.
  - add one or more points to make the input full dimensional.

nallath, 2017-04-24:
Thanks for reporting this!

We've noticed that for certain models a library that we use misbehaves (and crashes)

Ghostkeeper, 2017-04-25:
I'm unable to reproduce this so far. Are there specific models we need to load or does it happen with any model?

ianpaschal, 2018-02-27:
This was not reproducible and almost one year old, so I'm closing it.

closed 2018-02-27 by ianpaschal