Ultimaker/Cura #1730: (Still) Can't delete default materials in 2.5

ex-nerd, 2017-04-23:
It's nice to see the list of default/included materials grow with the release of v2.5, but the inability to delete them is a huge annoyance.  Here are some reasons why I might want to remove/suppress these files on my own:

* I will never be printing with ABS, so I don't want to have to scroll by all of the ABS materials every time I want to change to one I use.
* My printer uses an analog temperature sensor, so the manufacturer specified temperatures do not actually work for me and now I need to duplicate it and try to remember which is my working copy, and which is the non-working vanilla one (e.g. polyplus is configured for 210, but I need the printer set for at least 230 in order for it to print properly)
* There are some brands of filament that I just have no interest in using, and don't need to scroll past them every time I want to change my printer from one color to another.

I know that I can delete the files (which I have done), but it's seriously annoying to have to do this every time I upgrade the app.

Please provide a way to edit/override/delete/hide the default materials.

nallath, 2017-04-24:
I can see how this is annoying, but i fear that we won't implement this anytime soon. It's going to be quite a bit of work to get this working right and it's only going to be usefull for a few people. The people that would use it, are also not likely to be people that have Ultimakers (and as you might imagine, we can only spend our time once and UM related improvements will get more priority).

If someone feels like this is a cool feature, we're open for pull requests.

awhiemstra, 2017-04-24:
Note that deleting is going to be impossible, because these materials are located in locations that are either only writable with special permisisons (Windows, Linux from package) or completely read-only. (OSX, Linux AppImage) So the best we may be able to do is to hide them.

Ghostkeeper, 2017-04-25:
So our answer is then "We're not going to do this, but a pull request is welcome"?

dtwrv6, 2017-05-06:
Also can't delete materials I created in previous version 2.4 - they were invisible before and now the list of materials is way too big.  Please fix the problems with not being able to delete materials.

Ghostkeeper, 2017-05-10:
If they are materials you created yourself, they should be deletable. If that is not possible, that is a different issue. This issue is only about the default materials, which is fairly hard because the default materials are stored in a location on your hard disk where Cura can't write.

closed 2017-07-21 by BagelOrb