Ultimaker/Cura #1733: TweakAtZ flow

maukcc, 2017-04-24:
If you tweak the flow for extruder_1 for one layer, the reset to normal looks like:
M221 T1 S-1.000000
resulting in a flow of -1 %

this is of coarse not correct, it should be:
M221 T1 S100

Also the T1 is not supported by marlin firmware.

Ghostkeeper, 2017-05-01:
We hardly maintain TweakAtZ because it's awful code that we never should've ported. I'm pushing for a rewrite of the thing.

anton-veretenenko, 2017-05-21:
I can also confirm this behavior with tweaked speed values. Possibly other values as well.

Dim3nsioneer, 2017-06-09:
@Ghostkeeper So can we declare work on TweakAtZ stopped for now and you maybe take it out for 2.6?

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-23:
Yes, this still reproduces in Cura 3.0.4. Nothing is extruded after the tweak layer has completed printing.

Ghostkeeper, 2018-02-21:
This bug was removed from our planning because it was not seen as important enough.

mahtDFR, 2020-02-19:
There has been no activity on this issue for 30 days, so I am closing it.

Ghostkeeper, 2020-02-27:
Yes that's because this was deferred. I think you confused that with tickets that were waiting for a response?

closed 2023-06-13 by github-actions[bot]