Ultimaker/Cura #1734: Feature request: add layer view colour scheme to show speed

smartavionics, 2017-04-24:
S3D gives you the option of colouring the layer view based on the hot-end movement speed. It would be nice to have that capability in Cura.

Ghostkeeper, 2017-05-01:
What colour scheme would be appropriate for Cura? Brighter is faster?

We already use brightness to indicate the active extruder though.

fieldOfView, 2017-05-01:
I think @smartavionics suggests a third layerview mode: Line type, Material Color and Printing Speed. 

> We already use brightness to indicate the active extruder though.

That seems to be MIA since the new layerview work.

smartavionics, 2017-05-01:
Hi Guys. Yes, I am suggesting a new layer view mode.  S3D uses a rainbow from blue (slow), green, orange, yellow to red (fast). I don't like that personally and would go for a much simpler two colour scheme such as dark blue for slow and light yellow or orange or even white for fast. Although, thinking about that, half way between blue and yellow would be greenish but even so, I am advocating a simple colour scheme. Tastes vary so unless you give the user the option to select the colours at the limits, someone is always going to moan about the colour choice!

Ghostkeeper, 2017-08-08:
This was just rejected by our project manager because it was inactive for 3 months.

smartavionics, 2017-08-08:
OK, I'm not going to do it myself and if nobody else is likely to then let's close this issue.

smartavionics, 2017-11-14:
That sounds good. Even though the current layer view is quite pretty, it's still lacking a lot of functionality compared to other slicers. When I am working on cura  gcode issues, I currently view the gcode using the craftware slicer as a viewer. It's a little buggy but is so much better than cura for looking at gcode. Some useful stuff in there that could possibly be incorporated into cura. One of the very best features in craftware is the popup that displays all the info you need about the current line being printed (speed, extrusion rate, z, etc.) so you can scroll through the lines in the layer and see exactly what the gcode is doing at each line.

ChrisTerBeke, 2017-11-14:
He's building line-by-line simulation support including nozzle position, colors per flow rate (and/or print speed), colors by layer height (handy for my adaptive layer height feature).

diegopradogesto, 2017-11-15:
Yes, I started doing just the line by line view (#2713), but I had time enough for creating two new options for coloring the lines: Feedrate (aka. Speed) and Layer thickness. I will put some pictures and video.

smartavionics, 2017-11-15:
Hi, that sounds good. If you haven't played with it yet, I really recommend using the craftware slicer to view some gcode as it has some nice features you may want to look into. They also have an option to colour by extrusion rate and, as I mention above, their pop-up that includes all the info about the current line is very good. I am looking forward to seeing what you have done. Cheers.

diegopradogesto, 2017-11-15:
I was using CraftWare as an inspiration because I saw that you took some screenshots from that software. I didn't add the GCode information aside but it could be interesting.

smartavionics, 2017-11-15:
The scrolling gcode display is useful for me as a developer but I doubt if most people would need it. The little popup that shows the info for the current line is definitely worth recreating if possible (or something equivalent).

diegopradogesto, 2017-11-26:
This feature is implemented now in version 3.1

closed 2017-11-26 by diegopradogesto