Ultimaker/Cura #1736: M80 and layer 0 bed and nozzle temperatures

blole, 2017-04-24:
`M190 S{material_bed_temperature_layer_0}` and
`M109 S{material_print_temperature_layer_0}`
are not recognized as setting the bed/nozzle temperatures in the starting gcode and additional `M190`/`M109` commands are inserted before the starting gcode.

With
```ini
material_bed_temperature = 95
material_bed_temperature_layer_0 = 100
material_print_temperature = 230
material_print_temperature_layer_0 = 235
```

setting the start gcode to:
```gcode
M80
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
```
produces
```gcode
M190 S100
M104 S235
M109 S235
M80
M190 S100
M109 S235
```
Which won't work, since my build plate won't heat up without M80. :)

---

A workaround is to set the start gcode to
```gcode
M80
M190 S{material_bed_temperature}
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature}
M109 S{material_print_temperature_layer_0}
```
which produces
```gcode
M80
M190 S95
M190 S100
M109 S230
M109 S235
```

blole, 2017-04-27:
[StartSliceJob.py#L250-L251](https://github.com/Ultimaker/Cura/blob/37ec14c239d0dfcce9876632246b6cd6f962ae17/plugins/CuraEngineBackend/StartSliceJob.py#L250-L251)
```py
settings["material_bed_temp_prepend"] = "{material_bed_temperature}" not in start_gcode #Pre-compute material material_bed_temp_prepend and material_print_temp_prepend
settings["material_print_temp_prepend"] = "{material_print_temperature}" not in start_gcode
```
looks like the place.

Ghostkeeper, 2017-05-02:
Yeah, I've noticed this before.

I made a quick fix for you: f612a535636749f055cd54ca18f9db90c1c1b93b

This fix should become apparent if you can build this from source, or otherwise with the next public release we'll make, which will probably be the 2.6 beta in a few weeks.

closed 2017-05-02 by Ghostkeeper