Ultimaker/Cura #1737: One at a time and layer zero temperatures

blole, 2017-04-24:
When using the "one at a time" print sequence and a different build plate temperature for the initial layer, the head doesn't move away from the just finished part before setting a new build plate temperature and waiting for it to be reached, which can take several minutes. This leaves a molten mark on the part.

Additionally, the the layer zero nozzle temperature only has effect on the first part as it's not set again for subsequent parts.

Example generated gcode between switching parts
```gcode
...
;TIME_ELAPSED:497.206434
G1 F1500 E378.85954                 ; retract
M107                                ; fan off
M190 S105                           ; wait for build plate temperature
G92 E0
G0 F7200 X173.983 Y31.175 Z20.998   ; lift z
G0 X45.386 Y74.989
;LAYER_COUNT:79
;LAYER:0
G0 F3600 X44.596 Y69.818 Z0.3
;TYPE:SKIRT
...
```

Ghostkeeper, 2017-05-02:
Hmm, never occurred to me, but I can imagine that is a problem since we have the initial layer temperature. I can see how that gives bad results.

Ghostkeeper, 2017-09-18:
This was just rejected by our project manager for being inactive for too long.

blole, 2017-09-22:
That's a shame, it really bugs me. If you could point to where in the code I would go to add a gcode-line to either lift z or move to the beginning of the next part, I could have a go at it. :)

Ghostkeeper, 2017-09-26:
Sure. I suggest that you put it somewhere here: https://github.com/Ultimaker/CuraEngine/blob/896d6fbadad004f13049a7ebf7dbc90c03e9873a/src/FffProcessor.cpp#L101

Or maybe here, via a function in FffProcessor called something like "prepareNextMeshGroup" that only gets called between two mesh groups and not if there's only one mesh group: https://github.com/Ultimaker/CuraEngine/blob/896d6fbadad004f13049a7ebf7dbc90c03e9873a/src/commandSocket.cpp#L379

A "mesh group" is the term for a group of meshes that gets printed all-at-once. In one-at-a-time mode, every mesh is put in their own mesh group. Otherwise all objects get put in the same mesh group.

Ghostkeeper, 2017-11-06:
Fixed this here: https://github.com/Ultimaker/CuraEngine/commit/f32d2e3972cf9a17f921367a81868c605b22176a

closed 2017-11-06 by Ghostkeeper

blole, 2017-11-06:
Haha, great, thanks 😄