Ultimaker/Cura #1744: Cura window is too large for small screens

onitake, 2017-04-25:
This is a regression since Cura 2.3.0, and it affects all versions up to the latest HEAD. Cura 2.1.3 was working fine.

The main window of Cura does not scale beyond certain minimum dimensions. This is a serious problem on a low-resolution screen, as the window may be cut off by other desktop elements such as the task bar.

This happens to me on a machine with a resolution of 1366x768 pixels, where title bar, menu bar and task bar occupy 96 pixels vertically. This only leaves about 672 for the Cura main window contents.

Please remove this artificial restriction and make the window scale as it did before.

Ghostkeeper, 2017-05-02:
This seems like it's not a bug that would occur naturally. Someone did this on purpose.

I traced it back to this commit: https://github.com/Ultimaker/Cura/commit/e8743abe68da4403b361e3c9f91bd7a2f917eba5

I'm sure @fieldOfView had a reason to limit the window size. He'll have a word or two to say about that. If he agrees, I'll reduce the minimum window size a bit further.
There is the problem of the tool icons falling off the screen then, but that already happens right now anyway and we need a solution for that.

For now, a workaround for you is to modify the theme file to reduce the minimum window size.

fieldOfView, 2017-05-02:
I set _a_ minimum size so that the viewport width would not get smaller than 1 px wide. The minimum height I chose was so that all UI elements would fit vertically (most notably: toolbar + layerview dropdown + cura logo). I checked if this was workable on a WXGA screen (1280x800), but did not take screens with a 768 height into account. 

The minimum height can be reduced if we accept the Cura logo falling behind the layerview dropdown.

Note that with the addition of extruder buttons in the toolbar, more space may be required to fit all buttons into the window.

Ghostkeeper, 2017-05-02:
The logo already falls behind the layer view panel even on the default size, let alone the minimum size ;)

fieldOfView, 2017-05-02:
That likely means that part of the layerview ui will be unusable on 768p screens for the UM3. But that is less problematic than the SaveButton / MonitorButton area buttons being cut off.

onitake, 2017-05-02:
How about moving the layer view to the top of the viewport?
That would leave plenty of space.

onitake, 2017-05-03:
I found a workaround that may be useful to some users:
My window manager allows me to turn windows into fullscreen windows, which removes title and task bar while they are active. This leaves enough space for the layer view. You can also bind this action to a hotkey.

Ghostkeeper, 2017-05-05:
I modified the theme to have a slightly reduced minimum size.

Reordering buttons when the window is made smaller would be nice, I think. The layer view shouldn't be placed on top with normal-sized screens though.

onitake, 2017-05-06:
@Ghostkeeper Can you tell me which commit/patch set this was?
I'll just test the latest HEAD otherwise.
Thanks.

fieldOfView, 2017-05-07:
See https://github.com/Ultimaker/Cura/commit/d006db5b2c961bd2d517eb656163c255c62d1737. It is a very simple change.

I would advise against switching to head currently. A very large chunk of code just landed which causes some issues - at least on my install.

onitake, 2017-05-07:
Oh. Ok.

With this change, the main view is ok (including the logo), but the layer view is cut off.
I can still use if I switch the window to full screen mode.

Ghostkeeper, 2017-05-11:
We can't easily do something about the layer view panel. There are some new UI overhauls going around though that would (have to) take care of that.

closed 2017-05-11 by Ghostkeeper

jianqingHuang, 2018-07-15:
Traceback (most recent call last):
  File "/root/uranui/UM/Qt/Bindings/MainWindow.py", line 78, in __init__
    Application.getInstance().setMainWindow(self)
  File "/root/uranui/UM/Qt/QtApplication.py", line 384, in setMainWindow
    self.mainWindowChanged.emit()
  File "/root/uranui/UM/Signal.py", line 212, in emit
    self.__performEmit(*args, **kwargs)
  File "/root/uranui/UM/Signal.py", line 315, in __performEmit
    func(dest, *args, **kwargs)
  File "/root/curaui/cura/../plugins/PostProcessingPlugin/PostProcessingPlugin.py", line 266, in _createView
    Application.getInstance().addAdditionalComponent("saveButton", self._view.findChild(QObject, "postProcessingSaveAreaButton"))
AttributeError: 'NoneType' object has no attribute 'findChild'

fieldOfView, 2018-07-15:
That’s a nice traceback. If you want us to do anything with it, please use your words to describe what lead to this. We can’t help you otherwise.