Ultimaker/Cura #1745: The printing is stoping a few minutes after it started

Danielrug05, 2017-04-25:
Hi guys

Ijust installed the official versio  of CURA 2.5 and it is presenting an issue when i am printing, because it is stoping the printing a few minutes after it starts. The printing head justs stops and the nozzles temperatures drop to 0 (this is when i am printing with a two extrussion printer, just using one nozzle). why do you think this is happening?

PD: i am not using an Ultimaker, mine is custom.

Thanks.

nallath, 2017-04-26:
Could be a whole lot of things. Might be the printer, might be Cura. Without more information we can't be sure.

Danielrug05, 2017-04-26:
what kind of information do you need? i will try to get it.

fieldOfView, 2017-04-26:
What sort of printer do you have, which controller board, what firmware?
What OS do you use?
What do the logs say?

Ghostkeeper, 2017-05-02:
Perhaps include a sample of the g-code that causes it to stop, too, so that we can see whether Cura's output is wrong or not.

Danielrug05, 2017-05-02:
Hi to all, 

My OS is windows, the controller board and the firmware wasnt disclosed by the printer manufacturer and i dont have the logs. I tried to get this info but i couldn´t. 
Attached is the gcode you asked. With this one the printer stops but in the screen it appears like if the head continues to move, while the temperatures drops abruptly. 
[gcode failure.txt](https://github.com/Ultimaker/Cura/files/971544/gcode.failure.txt)

Ghostkeeper, 2017-05-04:
I see nothing in that g-code that should cause the print to stop.

* There is no pause command.
* There are temperature commands in there, but the only place where it sets the temperature to S0 is at the end. For the rest it alternates between S200 and S205. The bed temperature isn't changing much either. There are no big jumps in the temperature that could cause the print to wait very long because of wrong estimates.
* The acceleration and jerk seem normal to me. Some speeds are on the low side (like 10mm/s) but that could not be mistaken as freezing.

Maybe the PID-controller for the temperature on your Giga Box isn't handling all the minute temperature changes. You could amend that by making sure the stand-by temperature, initial layer temperature, final temperature and start temperature are all the same. For the rest, I don't know what you could try.

Danielrug05, 2017-05-04:
Hi,

Well, i did that. I adjusted all the temperatures to be the same as the printing, but at some point of the printing the problems appear. My printer worked well when it was single extrusion with all the versions of CURA. Then i installed a dual extrusion kit and the manufacturer updated the firmware to manage the dual extrusion, i think thats when the problems started. I can only print with CURA 15.04.6, but the times are to long compared to the ones i could print with the lastest CURA. As i said before i have a custom FDM printer and i managed to include it into the fotware by creating new definitions and extruders, based on some of the existing ones. 

Although, thanks for your help. I appreciate it very much.

Ghostkeeper, 2017-05-05:
I can't change or test anything for you since I don't see what's causing the problem.

One other thing you could try is to disable Acceleration and Jerk control. Perhaps your printer's firmware isn't handling that all too well.

fieldOfView, 2017-05-05:
@Danielrug05, how are you printing? Via USB or SD-Card? Something I we have not ruled out in this thread is - if you are using USB - that the connection may be dropped during the print.

fieldOfView, 2017-05-05:
Is there any reason you keep saying "the manufacturer", instead of letting us know the brand? Knowing what hardware you are using could be helpful in determining what's going on.

Danielrug05, 2017-05-05:
Hi, 
I am printing via SD-card. I keep saying the manufacturer because i am talking about the company and the dude who made it, my printer is a Protolab 3D and are made here in Colombia, the quality is awesome, but they are not as flexible as ultimaker, for example.

BagelOrb, 2017-07-23:
This seems to be a firmware issue.

This issue seems to be outdated.

We could reopen the issue if needed.

closed 2017-07-23 by BagelOrb