Ultimaker/Cura #1746: Cura 2.5.0 Crashing my Macbook Pro

rekom, 2017-04-26:
Hi, i have a macbook pro 2005. Since i've updated to Cura 2.5.0 i've got some artifacts on the user interface. It was really slow but i managed to slice my design. 
Strange thing, today it started to be really really slow, and crashed my computer. I had a lot of programs open and i thought i didn't have anymore memory left. 
Open it again, simple file to slice, nothing open, and it crashed again... Like computer frozen, had to hard reboot, never happened again. 

I'm running MacOS 10.12.4

rekom, 2017-04-26:
Ok uninstalling Cura and deleting Macintosh HD\Users\[user_name]\Library\Application Support\cura (which i didn't do my first attempts)

Fixed my problem.

Ghostkeeper, 2017-05-02:
So something in those settings was causing the problem.

We're not able to reproduce the issue here.

closed 2017-05-03 by nallath

Ageleia, 2017-05-21:
Thanks for the tip **reckom**! I had the same problem (mac book pro crushed once Cura started) and after defeating the suggested folder and reinstalling the software I can use it without a problem.