Ultimaker/Cura #1747: Inquiry: Is the 1st layer grid for support structures necessary?

filipgoc, 2017-04-26:
The 1st layer for support structures is... indeed different. 
- It does not respect XY distance
- it does not respect density (though there is a correlation)
- it does not respect the pattern (though there is a connection)

Why is this the case? 

I can see how something like this grid could improve adhesion, but it also adds a slew of disadvantages (lack of control, difficult removal...). 

Is there any way to disable this first layer that I'm missing? 

Or is it a critical feature I just don't appreciate? 

[image]

BagelOrb, 2017-04-26:
There's no way for a user to hanger that behaviour at the moment.

The pattern and density were indeed many for better adhesion, but i don't know where the xy distance comes from.

I don't really see the problem, though. Why would this be more difficult to remove?

filipgoc, 2017-04-27:
@BagelOrb 

> I don't really see the problem, though. Why would this be more difficult to remove?

### more difficult in two ways/cases that are unrelated
1. Depending on the model and material, sometimes I want to squish the 1st layer more in order to get supreme adhesion. Since I'm not able to adjust the XY distance of the support from the model (the setting only affects layers 2+), the support ends up fused with the model. 
2. A basic zig-zag infill (as generated by simplify3d for example) is easy to remove, because it can be squished like an accordion. the grid-like first layer prevents me from doing that.

BagelOrb, 2017-04-30:
1. I don't see how that is related to the first layer grid pattern. Please
explain. Also: take a look at the Minimum Support XY Distance setting.
2. I don't think that one layer is making support removal that much more
difficult. I don't think it's worth adding another setting in the interface
for.

Ghostkeeper, 2017-05-02:
I'm not seeing the problem described in 1. If I change the support X/Y distance to 20mm, the first layer also generates support 20mm away from where the model touches the floor.

Squishing like an accordion is not really viable for just one layer. Pulling it off all in one piece is, though. And I can imagine some other pattern could have advantages as well. If there are other settings that don't affect the first layer, I'd say that is a bug as well. You'd expect the first layer to have the same support settings as the rest, except the pattern.

filipgoc, 2017-05-02:
## add 1.  -  1st layer vs the rest of support
I suppose this is a bug then. 

here is a screenshot showing only the support visible with the XY distance exaggerated to 4mm. Cura 2.5, Mac OS Sierra. you can see that the XY distance has no effect on the 1st support layer. 
[image]

## add 2.  zip zag pattern 1st layer? 

This one is a matter of preference and personal experience I guess @BagelOrb. 

based on my experience with supports from Slic3r, Simplify3D, Cura, and Z-Suite, zig-zag harmonica is easiest to remove specifically because it's easy to squish in one direction, but it does not always make a difference. 

### case A
for a support like this, the first layer holding it together does not make much difference for the ease of removal, as you can still easily destroy the model-touching parts, or pull it off as one piece. 
[image]

### case B
however, for a support like this, it's truly great to be able to print a nice dense harmonica, and then use a lever to squish a part of the support and pull it out with pliers in pieces. the 1st layer support grid prevents that. 
(only support shown. you can imagine the part i'm sure.)

In fact, i read somewhere on ultimaker forums an advice for support removal, that was basically "cut the 1st support layer grid so that you can squish the rest of the support"
(sorry, didn't keep the reference) 
[image]

filipgoc, 2017-05-02:
@Ghostkeeper 
Got the culprit. Not sure if bug OR if UX confusion. 

The trouble is the XY/Z distance priority. 
The default behavior is  ```Z Overrides X/Y``` - which also houses the tricky case. 

👎  ```Z Overrides X/Y``` enabled and using ```support X/Y distance``` 
- the 1st layer grid never changes.

👍  ```Z Overrides X/Y``` enabled and using ```support X/Y distance``` 
- the 1st layer grid behaves as expected

👍  ```X/Y overrides Z``` enabled
- always behaves as expected

The tooltip for the function makes it sound like it's only applying to behavior around overhangs... so either the tooltip is a bit confusing or there is a bug. 
[image]

BagelOrb, 2017-07-21:
https://github.com/Ultimaker/CuraEngine/pull/567
It is being considered whether this fix should be done ATM

BagelOrb, 2017-07-21:
Yes it is need for adhesion, because support doesn't have much contact area with the build plate.

I'm sorry. This is not going to be fixed or at least not any time soon.

BagelOrb, 2017-07-23:
We could make a user setting to control the first layer becoming grid or not.

Ghostkeeper, 2017-10-30:
Our project manager just removed the issue (to control the first layer support pattern) from our planning because it's more than 12 weeks old.

samibaskir, 2018-01-31:
The problem with this grid is that, it is essentially a mindless grid, Your retraction setting have to be spot on to be able to build this grid cleanly, because there will be a lot of retractions! 

With some materials like PETG, it is really hard to build this grid cleanly, even if you have perfect retraction (because you can't run part fan in the first layer, and PETG will smear all over)

I believe it makes sense to keep the grid, but it has to be an interconnected grid, such that you can extrude it with the least number of retractions. It is very important to build this right because it is on the first layer. 

Also an option to choose a solid layer instead of the grid would be nice. It would provide better adhesion if your material does not have warping tendency

filipgoc, 2018-01-31:
Just in case someone finds this issue in the future who shares my dislike of the 1st layer grid: 

I have actually now switched to _cross pattern_ - this one does _not_ do the grid on the first layer, and it's close enough in airiness to zig zag to work as I like. 

Peace.

samibaskir, 2018-01-31:
It is not that I dislike the grid, it really does increase the stability of supports. I just point out that it is not currently being built in an optimal way, there are a lot of intermittent retractions, most of which is avoidable.

filipgoc, 2018-01-31:
(Indeed. My follow-up post was not for you, samibaskir. You just reminded me of this thing. Just a closure.)

Ghostkeeper, 2018-02-26:
Perhaps using the new Connect Infill Lines setting for grid would be an improvement then, but we'll first have to test that well...

QPS-makes, 2018-05-17:
The first layer support pattern should be controlled by users. The default grid significantly increases the number of retractions and can cause failures in the first layer for narrow support areas, which a zig-zag pattern would avoid in most cases. I hope the next version of Cura has a fix for this bug.

compatibilizer, 2018-05-17:
The grid is a very dependable base for the support structure. Having no base, it will have a tendency to lift off from the build plate. The problem is not the grid, but the way it's built.

Elocai, 2018-05-22:
Need to increase the density of the first layer to create more adhesion, pls add a setting

alekseisasin, 2018-07-02:
Hi @Elocai , I think, if cura will let you define **Initial Layer** Support Distance, it will be enough to improve adhesion.

ianpaschal, 2018-07-02:
Just reviewed and tested @alekseisasin's new setting. This will be in the next version of Cura.

ianpaschal, 2018-07-02:
[image]
[image]

closed 2018-07-02 by ianpaschal

Elocai, 2018-07-02:
great to see it implented thx : )

zerox1212, 2021-02-11:
Hate to resurrect this, but is there a setting for this yet?

Printing PETG with so many retractions is huge pain. The grid isn't even needed for PETG because it already sticks to the build plate so strongly removal is difficult.

It also adds a significant amount of print time if you are doing a large part!

Ghostkeeper, 2021-02-19:
There's a line distance setting and an extruder setting, but indeed no setting to change the pattern for support on the first layer.

compatibilizer, 2021-02-19:
What if we kept the pattern, but try to print it with the least number of retractions? It works well for the infills.

zerox1212, 2021-02-19:
@compatibilizer I would prefer to have the option to just turn off the grid. It wastes time and filament, especially when I'm printing big stuff like on CR10 Max.

compatibilizer, 2021-02-19:
I find it useful for those odd supports, when you need to support a small feature at a high Z layer (ie. very long and thin support structure)