Ultimaker/Cura #1748: Text vs Label

fieldOfView, 2017-04-26:
Recently a number of changes were made to change Label components into Text components. I have yet to find the logic in which components were changed and which were not. Eg in https://github.com/Ultimaker/Cura/commit/2d14052f4e41f4346aa429439d46fe68182e9395 5 Labels were changed into a Text, but the 6th (which is a direct sibbling) was not.

Unfortunately the change of Label to Text in the parts of the interface mean that the text now looks asif it is rendered in another font:
[image]
If you look at the full sized screenshot, you will see that the texts below the "Connect via Network" button looks different than all the other text in the window.

It turns out the text is not rendered in another font, but using a different font rendering. This is how PyQt defines a Label:
https://github.com/pyqt/python-qt5/blob/master/PyQt5/qml/QtQuick/Controls/Label.qml#L79
The thing that sets the Label apart from Text are the color and the renderType. By using Text instead of Label, the texts are no longer rendered as Text.NativeRendering (but as Text.QtRendering instead).

Isn't the whole idea of the unthemed dialogs that they *do* use native rendering to fit in with the system theme?

I would argue that it is much better to use the same text rendering for all elements, or at least for all elements in one "theme".

The commit references "ugly fonts". Is there a screenshot somewhere?

nallath, 2017-04-26:
In some cases the rendering seems to fail for some operating systems. I don't have any screenshots, but the letters get warped somewhat.

fieldOfView, 2017-04-26:
I think I have seen what you describe, but this could always be fixed by making sure Labels were positioned on whole pixel positions. I'll look for an example to confirm we're talking about the same thing.

awhiemstra, 2017-04-26:
More important in my opinion, Text does not use platform style at all. This results in the configuration dialog now looking like this on my machine:

[image]

fieldOfView, 2017-04-26:
Ah, yes I was afraid that might be the case too, since Label specifically sets the (default) color to the system color.

Here's an occurance of "ugly font rendering", is that what we're talking about? See the second column:
[image]
It happens when resizing the Machine Settings window to an uneven number of pixels wide. What happens then is because the columns relate to half the parent window width, the second column ends up on a half pixel offset (uneven number / 2), and native text rendering does not like that.

Instead of fixing that by changing the font rendering to something that looks different than native rendering, I can also fix that by making sure the second column never ends up on a half pixel offset by rounding the width of columns to a full pixel.

Appesteijn, 2017-04-28:
As we found that on some Macs (usually, but not always) with low DPI the font redering was very ugly (just like the text in the screenshot above). We then found out it was the Label that used the system-font and the Text that used the packed font. I then tried to fix all occurrences that were found during testing 2.5-beta, by changing that Labels into Texts. The Labels that didn't give any issues, I left untouched.
Won't it be a pain to make sure all texts are rendered on pixels like in your branch?

nallath, 2017-07-27:
This has been fixed in the pull requests. Rejoice.

closed 2017-07-27 by nallath