Ultimaker/Cura #1750: A single filament cost is shared by all materials

ex-nerd, 2017-04-27:
With Cura v2.5 (macos), the material cost is shared across all of my custom material types.  Change it in one, and it changes all of them.

fieldOfView, 2017-04-27:
That's because you created the materials by duplicating another material. For details, see here:
https://github.com/Ultimaker/Cura/issues/1104#issuecomment-296591164

I hope I can make a fix for the confusion before Cura 2.6 goes into beta.

ex-nerd, 2017-04-27:
Gotcha, thanks.  So less of a GUID and more of a GNUID (globally non-unique id).  :)  I'll go update my files for now..

fieldOfView, 2017-04-27:
SGUIDbydi(biinyf): Shared Globally Unique ID because you duplicated it (but it is not your fault).

But that is a bit too long, so we settled on GUID.

fieldOfView, 2017-04-30:
See #1757

ex-nerd, 2017-04-30:
Out of curiosity, what's the user benefit of having multiple separate materials share an id?

fieldOfView, 2017-04-30:
The original idea was that there are protected/readonly profiles for materials. If a user thinks "but I always want to print mine a bit hotter", he'd make a copy of the material and edit the value.

ex-nerd, 2017-04-30:
Right, but when he/she makes a copy..  it's not actually a copy.  However, I think the proposed changes (especially if accompanied by UI changes to highlight shared fields) should go a long way to clarifying how things work under the hood.

ianpaschal, 2018-05-29:
I'm closing this and marking it a "won't do" because we're working on heavily redesigning the materials view. We're keeping this feedback in mind, but won't address it specifically until we're got a prototype of the new layout.

closed 2018-05-29 by ianpaschal