Ultimaker/Cura #1753: cura 2.5, profile seems to save incomplete

UltiArjan, 2017-04-28:
[image]
I've seen it before, but just tried to save a profile again, and after I save I still get a "star" behind the profile name, indicating there's been a change..... which is not posible as I just saved it....

enclosed the project file...
[MFU_LEGOwheel-v02.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/964163/MFU_LEGOwheel-v02.curaproject.3mf.zip)

Ghostkeeper, 2017-05-04:
Can't test this in master, since loading project files just broke yesterday :(
I'm not able to reproduce this by reproducing the settings as can be seen in the 3MF file with a text editor. I must be missing something. I guess we'll re-test once loading project files is repaired.

BagelOrb, 2017-07-23:
@Ghostkeeper please retest

closed 2018-03-02 by ChrisTerBeke

Ghostkeeper, 2018-03-28:
We won't consider this a bug then because the definition file contains syntax errors. Please correct that definition before using it in Cura!