Ultimaker/Cura #1754: Cura 2.5 Support doesn't touch raft

rs4race, 2017-04-28:
When slicing using rafts, supports, and 0 raft air gap there is a 1 layer gap between support structure and raft and no gap between part and raft. 

If I remove the raft there is no skipped layer.

Is there a setting I am missing?

Ghostkeeper, 2017-05-04:
I'm not seeing this, but I am seeing something much stranger: It's printing 6 support layers, then 6 mesh layers, then the first 6 support layers again at the same height as the original, crashing into the first 6 support layers. Something is up.

rs4race, 2017-05-08:
It appears that if I set the initial layer height to 0.2 mm there will be an air gap between support and raft, but no gap between part and raft. This is with raft air gap set to 0 mm. It's as if the first layer of the support is skipped.
If I change the initial layer height to 0.25 mm the first layer of support is now generated! The part is touching the raft and the support is touching the raft.

spilz87, 2017-07-07:
i'm using 2.6.1 and I still have the same issue : 
the first layer after the raft does not include support part, what ever height is initila layer and what ever  distance support / top layer is set ...
I don't really understand the logic of this, for me the support in contact with the raft should be stronger than the contact raft / piece 

this issue did failled several prints :(

waiting for a solution

Ghostkeeper, 2017-07-10:
We have a solution in 2.7, but that didn't make it to the 2.6 release.

yufanyufan, 2017-07-10:
What is the working around before 2.7?

spilz87, 2017-07-10:
it's a good news for 2.7
if there is a known solution with multiple of layer height, it will be fine to explain it :)

Ghostkeeper, 2017-07-11:
A workaround is to set the initial layer height to 0.0001mm. It'll introduce some odd travel moves but the support will be at the same height as the model then.

BagelOrb, 2017-07-21:
I cannot reproduce this any more.

What's your initial layer height?
Easier yet: attach a project file which causes the problem.

Ghostkeeper, 2017-08-02:
The 2.7 beta is being released today. Perhaps you can test again?

There should be a blog post on Ultimaker's website somewhere today, but until then, some of our regulars will know where to download Cura *open* betas.

ianpaschal, 2018-05-26:
Can't reproduce and more than one year old, so closed unless we hear otherwise.

closed 2018-05-26 by ianpaschal