Ultimaker/Cura #1756: Inaccurate g-code

ghsamme, 2017-04-29:
Hi

I get gcode that isn't what it should be. After mooving the model on the build plate a bit it is correct. Se attached before/after screen shots. The model is a flat thin box with 8 pillars on. I've seen this problem on many models.
Complete model before move:
[image]
Complete model after mobe (correct code):
[image]
Model before move (surface should be flat):
[image]
Model after move (correct code):
[image]

Kind regards
Samuel Starkenberg

smartavionics, 2017-04-29:
Hi, could you please share the model so I can have a play with it?
What layer height are you using?

ghsamme, 2017-04-29:
I use 0.2 mm. Here are the stl file, the gcode file and I also exported the cura profile.
[Pillars.zip](https://github.com/Ultimaker/Cura/files/966151/Pillars.zip)
[pla profile.zip](https://github.com/Ultimaker/Cura/files/966152/pla.profile.zip)

smartavionics, 2017-04-29:
Hello Samuel. I cannot reproduce this with Cura 2.5 on a Linux box. I have to turn the model through 90 deg after loading but then it appears to slice just fine. Moving it around on the build plate makes no difference.

smartavionics, 2017-04-29:
I tried a bit harder and did manage to reproduce the problem. The difference was that I was originally using a first layer height of 0.2 but your profile uses 0.3. With 0.3, it behaves like your images show.

Anyway, as far as I can tell, the problem is down to the fact that the base rectangle of your model is not of uniform thickness. The left hand side appears to be lower than the right and so for certain combinations of layer thicknesses one less layer is used on the left hand side compared to the right.
Why it changes by moving the model I do not know but I guess if the rectangle thickness is really close to the boundary for N or N+1 layers then moving the model will slightly alter the position of that boundary?

Ghostkeeper, 2017-05-05:
I have a hunch of what might be happening here. It appears that the slice of that layer is including only part of the top surface. And it looks like the border of that top surface is exactly on the X=0 plane.

We know that there are minute inaccuracies in the rotation in the front-end due to the limited accuracy that a computer can store in a float, for the transformation matrix. If you had to rotate it around the Y axis, this would explain the difference in how it's sliced. Otherwise it could just be an edge case where the plane falls exactly on a slice height. I suggest moving the object to the side of the build plate. It works there.

BagelOrb, 2017-07-21:
I don't think this is a bug we can ever solve. It's just an edge case where some layer exactly aligns with a slice layer.

closed 2017-07-21 by BagelOrb