Ultimaker/Cura #1758: Cura not do slicing

userosos, 2017-04-29:
Cura 2.5 not do slicing. I add some model into cura and press slicing before it i can see that cura do slicing very long time. I have Debian 9 stretch. In the termianl i can see:
`UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
UM.Qt.QtApplication.__init__ [75]: Detected most suitable OpenGL context version: 4.1 Core profile
2017-04-29 23:27:04,018 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ConsoleLogger
2017-04-29 23:27:04,031 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-04-29 23:27:04,032 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraEngineBackend
2017-04-29 23:27:04,033 - WARNING - UM.PluginRegistry.loadPlugin [74]: Plugin ConsoleLogger was already loaded
2017-04-29 23:27:04,037 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LocalFileOutputDevice
2017-04-29 23:27:04,039 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin WireframeView
2017-04-29 23:27:04,041 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SimpleView
2017-04-29 23:27:04,044 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CameraTool
2017-04-29 23:27:04,046 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SelectionTool
2017-04-29 23:27:04,050 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin MirrorTool
2017-04-29 23:27:04,053 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin TranslateTool
2017-04-29 23:27:04,056 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ScaleTool
2017-04-29 23:27:04,060 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin RotateTool
2017-04-29 23:27:04,062 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin FileLogger
2017-04-29 23:27:04,064 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin OBJWriter
2017-04-29 23:27:04,068 - WARNING - STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2017-04-29 23:27:04,069 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin STLReader
2017-04-29 23:27:04,071 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin OBJReader
2017-04-29 23:27:04,075 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin STLWriter
2017-04-29 23:27:04,080 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin AutoSave
2017-04-29 23:27:04,085 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin PostProcessingPlugin
2017-04-29 23:27:04,089 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ChangeLogPlugin
2017-04-29 23:27:04,094 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin XRayView
2017-04-29 23:27:04,099 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SolidView
2017-04-29 23:27:04,116 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin USBPrinting
2017-04-29 23:27:04,123 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFWriter
2017-04-29 23:27:04,126 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-04-29 23:27:04,127 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 293, in _findPlugin                                                                                                                     
2017-04-29 23:27:04,127 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-04-29 23:27:04,128 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3.5/imp.py", line 244, in load_module
2017-04-29 23:27:04,129 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-04-29 23:27:04,129 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3.5/imp.py", line 216, in load_package
2017-04-29 23:27:04,129 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-04-29 23:27:04,129 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-04-29 23:27:04,130 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-04-29 23:27:04,130 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 673, in exec_module                                                                                                                                  
2017-04-29 23:27:04,130 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed                                                                                                                             
2017-04-29 23:27:04,131 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/cura/plugins/3MFWriter/__init__.py", line 6, in <module>                                                                                                                                  
2017-04-29 23:27:04,131 - ERROR - UM.Logger.logException [78]:     from . import ThreeMFWriter
2017-04-29 23:27:04,132 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWriter.py", line 11, in <module>                                                                                                                            
2017-04-29 23:27:04,132 - ERROR - UM.Logger.logException [78]:     import Savitar
2017-04-29 23:27:04,132 - ERROR - UM.Logger.logException [78]: ImportError: No module named 'Savitar'
2017-04-29 23:27:04,132 - ERROR - UM.PluginRegistry._populateMetaData [249]: Could not find plugin 3MFWriter
2017-04-29 23:27:04,134 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFWriter
2017-04-29 23:27:04,135 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-04-29 23:27:04,135 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 293, in _findPlugin                                                                                                                     
2017-04-29 23:27:04,135 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-04-29 23:27:04,136 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3.5/imp.py", line 244, in load_module
2017-04-29 23:27:04,136 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-04-29 23:27:04,136 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3.5/imp.py", line 216, in load_package
2017-04-29 23:27:04,136 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-04-29 23:27:04,137 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-04-29 23:27:04,137 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-04-29 23:27:04,137 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 673, in exec_module                                                                                                                                  
2017-04-29 23:27:04,137 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed                                                                                                                             
2017-04-29 23:27:04,138 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/cura/plugins/3MFWriter/__init__.py", line 6, in <module>                                                                                                                                  
2017-04-29 23:27:04,138 - ERROR - UM.Logger.logException [78]:     from . import ThreeMFWriter
2017-04-29 23:27:04,138 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWriter.py", line 11, in <module>                                                                                                                            
2017-04-29 23:27:04,138 - ERROR - UM.Logger.logException [78]:     import Savitar
2017-04-29 23:27:04,138 - ERROR - UM.Logger.logException [78]: ImportError: No module named 'Savitar'
2017-04-29 23:27:04,148 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin XmlMaterialProfile
[... cut ...]
2017-04-29 23:27:31,243 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [545]: Backend quit with return code 0. Resetting process and socket.
2017-04-29 23:27:31,281 - DEBUG - CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [310]: Sending slice message took 0.14105916023254395 seconds
2017-04-29 23:27:35,672 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-04-29 23:29:04,672 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

`

nallath, 2017-05-01:
As far as I can see you seem to be running it from source. You also need to install libSavtiar.

userosos, 2017-05-02:
In debian repo have not got the package. I installed the cura from PPA repo (https://launchpad.net/~thopiekar/+archive/ubuntu/cura).

nallath, 2017-05-03:
The PPA is not a "official" release from us. Its maintained by @thopiekar.

If you want everything to work, i'd recommend downloading the appimage.

thopiekar, 2017-05-03:
I personally see that the module is inside the package..
https://launchpadlibrarian.net/317966410/buildlog_ubuntu-zesty-amd64.libsavitar_2.5.0~201705021151~rev52~pkg144~ubuntu17.04.1_BUILDING.txt.gz

Never used 3MF before, just made the package. Make sure you've installed the latest updates from the PPA.

thopiekar, 2017-05-03:
Double checked whether Savitar importable here and yes, the 3MFReader loads fine without any blame of a missing module.
Thus there is something wrong with your installation.

userosos, 2017-05-03:
Hm...
cat /etc/apt/sources.list.d/thopiekar-ubuntu-cura-artful.list 
#deb http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial main
#deb-src http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial main
#deb http://ppa.launchpad.net/thopiekar/cura-master/ubuntu zesty main 
#deb-src http://ppa.launchpad.net/thopiekar/cura-master/ubuntu zesty main 
deb http://ppa.launchpad.net/thopiekar/cura/ubuntu zesty main 
deb-src http://ppa.launchpad.net/thopiekar/cura/ubuntu zesty main

It is correct?

thopiekar, 2017-05-03:
Maybe, I don't have an Debian installation here.
You should look for the Ubuntu distribution that fits to your Debian version the best.
I'm not totally sure which packages in sum have to fit, but libstd6 for example (could also be called differently) has to fit. Otherwise installing libsavitar won't work.
However, if you don't​ get any errors because of unsolvable errors in APT then everything should be fine.

However, you are experimenting with Debian over here and giving people support on Ubuntu is already enough for me. If you find a solution for the issue or can reproduce it on Ubuntu itself. Just let me know.

closed 2017-07-05 by thopiekar