Ultimaker/Cura #1759: include our printer in source code

nbennett1978, 2017-04-29:
Hi,
we are manufacturer of ALYA, a FFF printer for kids and youngsters. I'd like to ask to include the printer definitions into the source code. I tried to do that myself, but couldn't due to user permissions.
[alya3dp.def.json.txt](https://github.com/Ultimaker/Cura/files/966412/alya3dp.def.json.txt)

nbennett1978, 2017-05-01:
We used legacy cura (15.04) up until recently. The printer definition file i sent you was created with the generator link on cura's github repo- what exactly did you mean with your recommendation to update?

fieldOfView, 2017-05-08:
This is fixed now that #1761 has been merged. The Alya 3d printer will show up in the list of printers when adding a new printer in version 2.6 and up.

closed 2017-05-08 by nallath