Ultimaker/Cura #1767: Cura 2.5 wont start on OSX 10.8.5

Toby19, 2017-05-01:
Hi
I downloaded and tried to start version 2.5 - nothing happens
I started /Applications/Cura.app/Contents/MacOS/cura from command line.
This starts and I get a lot of debug. In the end the GUI appears but in the terminal I see an infinite loop:
2017-05-01 12:29:55,441 - DEBUG - UM.Backend.Backend._onSocketError [194]: Socket was unable to bind to port, increasing port number to 50811
2017-05-01 12:29:55,545 - DEBUG - UM.Backend.Backend._createSocket [209]: Previous socket existed. Closing that first.
2017-05-01 12:29:55,550 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [223]: Attempting to kill the engine process
2017-05-01 12:29:55,554 - DEBUG - UM.Backend.Backend._createSocket [209]: Previous socket existed. Closing that first.
2017-05-01 12:29:55,565 - DEBUG - UM.Backend.Backend._onSocketError [194]: Socket was unable to bind to port, increasing port number to 50812

You see the port number increasing on every iteration.
I can start the CuraEngine from command line:
/Applications/Cura.app/Contents/MacOS/CuraEngine slice -j /Applications/Cura.app/Contents/Resources/resources/definitions/ultimaker2.def.json -l test.stl

Any ideas why Cura can't start or connect the engine?
Thanks,
Toby

nallath, 2017-05-01:
This is usually because of firewalls / virus scanners blocking local connections.

Toby19, 2017-05-01:
Thank you for your suggestion. I didn't install a Firewall and as far as I know, there is no one out of the box in OSX 10.8.5.
Cura Version 15.06.03 runs fine and I see a connection between cura and CuraEngine. I don't think anything on my box limits sockets/connections (apart from privileged ports < 1000).

Toby

Ghostkeeper, 2017-05-05:
Perhaps it isn't finding the back-end properly, even?

You can set the location of the back-end by modifying your configuration file for Cura under /Library/Application Support/cura/cura.cfg. You'll want to add something like this:

    [backend]
    location = /Applications/Cura.app/Contents/MacOS/CuraEngine

This shouldn't do anything normally, but perhaps it helps if the engine can't be found.

Toby19, 2017-05-06:
Hi Ghostkeeper

I tried to use the configuration setting. No effect. To make sure that the file was read and used I started cura with dtruss (similar to truss/strace/..). AFAIK you have to be root to use dtruss. I found out that after the first start of Cura $HOME/Library/Application Support/cura/cura.cfg is created and read. So I added the configuration setting as well to the cfg in ~/Library/...
In the dtruss output I found (Cura running as root):
stat64("/var/root/Library/Application Support/cura/cura.cfg\0", 0x7FFF5EA45890, 0x101400AE8)             = 0 0
open("/var/root/Library/Application Support/cura/cura.cfg\0", 0x1000000, 0x1B6)          = 6 0
=> ok, cfg file is found and opened

read(0x6, ".....[backend]\nlocation = /Applications/Cura.app/Contents/MacOS/CuraEngine\n\n[gcoderead", 0x2000)              = 269 0
=> the proposed setting is read

Could you briefly describe in which order is 
- the socket created
- the client process CuraEngine starts
- what is server and what is client?

From my findings so far I would expect that Cura
- opens the socket
- binds to a "random" port (== server)
- starts the client with the bound port as parameter
 
I think the error is indeed related to the bind. In the dtruss log are several sockets created and used. When Cura is stuck in the infinite loop, I see 3 consecutive lines:

socket(0x2, 0x1, 0x0)            = 20 0
which means regarding to "man socket" and sys/socket.h
2 == Inet socket
1 == streamed
0 == IP
bind(0x14, 0x624F5BC70, 0x10)            = -1 Err#49
close(0x14)              = 0 0

Please don't get confused, the resulting file descriptor is in decimal, for the bind it's in hex. 
What is the content of the second bind parameter?
I tried to debug using gdb /Applications/Cura.app/Contents/MacOS/cura
waited to get Cura stuck and set a breakpoint at bind
I see the stacktrace but not the arguments. Probably the lib is stripped of debug info?

#0  0x00007fff8d90f9dc in bind ()
#1  0x000000010110c492 in Arcus::Private::PlatformSocket::bind ()
#2  0x0000000101105da8 in Arcus::Socket::Private::run ()
#3  0x0000000101107ea4 in _ZNSt3__114__thread_proxyINS_5tupleIJZN5Arcus6Socket6listenERKNS_12basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEEEiE3$_1EEEEEPvSE_ ()
#4  0x00007fff94394772 in _pthread_start ()
#5  0x00007fff943811a1 in thread_start ()

I think I have to search somewhere in the python part.
Which branch of Uranium does match Cura 2.5, probably 2.5?

I will continue digging, if you have an idea where to start, please let me know.

Toby

Toby19, 2017-05-06:
Hi again
I made a little progress.
I copied https://github.com/Ultimaker/Uranium/blob/2.5/UM/Backend/Backend.py
into 
/Applications/Cura.app/Contents/Resources/lib/python3.5/UM/Backend/Backend.py
 and replaced the last line with
        self._socket.listen("127.0.0.1:" + str(self._port), self._port)
(I'm not an experienced python programmer. May be I'm doing something horribly wrong)
The port is now part of the address.
I have now cura and CuraEngine connected via TCP.
Unfortunately if I load a stl file, it doesn't appear in the gui. Build time and cost rest at 0:00. I don't know if the gui has passed the stl content and/or file name to the engine.

Toby

Toby19, 2017-05-07:
Tada, I got it working. I could slice a model and write a gcode file.
My problem is that the file open dialog does return an empty list.
In /Applications/Cura.app/Contents/Resources/resources/qml/Cura.qml

    FileDialog
    {
        id: openDialog;

there an is iteration through the list of selected files:

  ```
for(var i in fileUrls)
            {
                Printer.readLocalFile(fileUrls[i])

```
This throws an error because fileUrls is empty regardless if I select one or multiple files.
I just called the method directly outside the loop with an hard coded name:
`Printer.readLocalFile("file:///Volumes/data/homes/toby/test.stl")`
This model is shown and sliced!

I would be very grateful if someone could give me advise how to get the selected file list. I tried to use pprint of python to inspect/dump the dialog instance. Qt doesn't like the import statement. Or is the problem in the parent class Dialog.qml?

Toby

Ghostkeeper, 2017-05-10:
In the `onAccepted` of that `openDialog`, perhaps you can add the line:

    print("=======================" + fileUrls)

somewhere. Then open Cura from the command line and see if that shows up anywhere, and if there is anything in the original fileUrls gotten from Qt. Perhaps do the same but with `fileUrl` rather than `fileUrls`.

Toby19, 2017-05-10:
Thank you for reading.
I added both fileUrls and fileUrl with a print as suggested:

            print("======================= fileUrls" + fileUrls)
            print("======================= fileUrl" + fileUrl)

and got:

qml: ======================= fileUrls
qml: ======================= fileUrl

So both are empty. 
It's not that the dialogs don't work at all.
The "save" dialog for the generated gcode works and creates a file on the disk. 
It is something in the open dialog. I tried to use 
console.trace
to get the stack and find out why the list is empty.
No success, I don't get a stack.

Toby

Ghostkeeper, 2017-05-12:
> It's not that the dialogs don't work at all.

Well, that `fileUrls` is as close as we can get to the original data. It's a property of the QML element, the source, the point where the data transfers from Qt to our code. So if that doesn't get filled, it seems that the load dialogue isn't working for us ;)

I'm at a bit of a loss. One thing that could be tried is to switch the type of dialogue. Qt offers a bunch of variants: Modal, not modal, multi-file, single-file, folder based, etc. The way they work is unfortunately very platform-dependent.

I don't know how to start debugging that however.

sbgraphic, 2018-02-15:
Hello,
as stated in #2537, I have a similar issue. Cannot launch Cura 2.5 to 3.2 ? any news about this ?
Best,
Eric

Ghostkeeper, 2018-03-08:
Is there anything in your log file?

ianpaschal, 2018-05-26:
Due to non-reproducibility and needing info for more than a year, this deserves to be closed.

closed 2018-05-26 by ianpaschal