Ultimaker/Cura #1768: Imported profiles issue (Cura 2.5.0)

kr15uk, 2017-05-01:
Ladies and gents,

I've found a bug with imported profiles. (profile imported from Cura 2.4)

Imported profile doesn't auto update automatically calculated fields, specifically temperature.
[image]

2nd glitch is that imported profile "takes over" profile view in settings.
I have active newly created profile in v2.5.0 but it shows partial data fields of imported profile instead of active/created one... Quite weird.
[image]

Thanks.

Ghostkeeper, 2017-05-05:
The profile you imported overrides the printing temperature, sets it to a fixed value. It doesn't scale along with the default printing temperature because the profile says it should be 195. You can click on the *i* to revert it to scale along with the default printing temperature.

The second glitch seems to be a bug indeed. I'll report that one in our internal tracker so we can plan it.

BagelOrb, 2017-07-23:
Is that second bug fixed now?

Ghostkeeper, 2017-08-04:
Nope, still in the backlog: CURA-3764

Ghostkeeper, 2017-11-06:
Well not any more. Our project manager just removed this from our planning because it's older than 12 weeks.

digitalfrost, 2022-08-08:
Stale ticket.
Refers to Cura 2.5
Was removed from backlog in 2017.
Can be closed?

closed 2023-10-18 by saumyaj3