Ultimaker/Cura #1773: Wrong build size for UM3

Dim3nsioneer, 2017-05-01:
Official build size of the UM3 is 215x215x200mm (single extrusion) and 197x215x200mm, respectively. Cura 2.5.0 allows only for 208x208x200mm and 190x208x200mm (build plate adhesion off). Exclusion zones for the clips etc. will reduce those sizes further.
I tested the dual x width of 197mm by changing the json file and it works with the official number.
Please correct the corresponding json files for the next Cura release.

nallath, 2017-05-02:
You always have max build volume. But having a max build volume does not mean that you are able to get the maximum model volume. 

Getting the max model volume will have a big impact on model quality. You will need to change a whole bunch of settings that can mess up your prints big time, unless you know what you are doing. Adding an option for it suggests that it's something that is safe to do, which it isn't. 

As for the missing mm, this can be caused by rounding. There is a bit of that going on in a few places. The end result can then be that you "loose" a milimeter. To be honest, i think we have more important issues to fix then getting an extra mm of build space. Especially as it's not an easy fix.

Dim3nsioneer, 2017-05-02:
Thanks for your reply. As from a technical point of view, the issue can be closed then.

closed 2017-05-02 by Dim3nsioneer